purduesigbots/pros-cli #6: TypeError: unhashable type: 'ListPortInfo'

edjubuh, 2018-09-05:
https://sentry.io/purdue-acm-sigbots/pros-cli-3/issues/675639322/

```
TypeError: unhashable type: 'ListPortInfo'
(2 additional frame(s) were not displayed)
...
  File "click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "pros/cli/upload.py", line 139, in ls_usb
    ports = find_v5_ports('system')
  File "pros/serial/devices/vex/v5_device.py", line 45, in find_v5_ports
    user_ports = set(user_ports) - set(system_ports)

TypeError: unhashable type: 'ListPortInfo'
```

Octogonapus, 2018-09-05:
For reference, this is the macOS Mojave issue I ran into.

closed 2018-09-08 by HotelCalifornia

edjubuh, 2018-09-08:
Closed by 5b649a7f1a2f372c84e61b730e03ead1c96779c5