purduesigbots/pros-cli #7: Won't install

SirCe, 2018-09-08:
#### Expected Behavior:
Upload the program that I wrote on my other machine with a different error
#### Actual Behavior:
Can't even install when I get messages telling me "abort/retry/ignore"  Retry does nothing.  And ignore goes to the next one.  I went through 400+ before aborting.
#### Steps to reproduce:
Install pros for classic
#### System information:
Operating System:  (Windows, OS X, Ubuntu)
Windows 10
PROS Version: (run `pros --version`)
cant run that if it's not installed
#### Additional Information

#### Screenshots/Output Dumps/Stack Traces
Here is the error from the other machine output in the terminal:

  Tether: Unknown tether connection (ConnectionType.unknown)
Joystick: F/W  w/ 0.00V
  Cortex: F/W  w/ 0.00V (Backup: 0.00V)
Unable to determine system type. It may be necessary to press the programming button on the programming kit. Continuing is usually safe. C
ontinue? [Y/n]: Aborted!
make: *** [Makefile:36: upload] Error 1

edjubuh, 2018-09-08:
If you're installing for all users, files are typically extracted to `C:\ProgramData\pros...\` and installed to `C:\Program Files\PROS` and you never have an issue with path length. It's possible that something else is going on or you've changed those paths.

What did you do to see this message:
```
Tether: Unknown tether connection (ConnectionType.unknown)
Joystick: F/W w/ 0.00V
Cortex: F/W w/ 0.00V (Backup: 0.00V)
Unable to determine system type. It may be necessary to press the programming button on the programming kit. Continuing is usually safe. C
ontinue? [Y/n]: Aborted!
make: *** [Makefile:36: upload] Error 1
```

From my understanding, you do not currently have PROS installed at all.

SirCe, 2018-09-08:
That error was on my first machine.

I didn't change the path.  No I do not have PROS installed at all right now.

3038922, 2018-09-09:
[image]
If I want to write code on another computer, do I need to modify directory?

edjubuh, 2018-09-22:
So the output:
```
Tether: Unknown tether connection (ConnectionType.unknown)
Joystick: F/W w/ 0.00V
Cortex: F/W w/ 0.00V (Backup: 0.00V)
Unable to determine system type. It may be necessary to press the programming button on the programming kit. Continuing is usually safe. C
ontinue? [Y/n]: Aborted!
make: *** [Makefile:36: upload] Error 1
```
is unrelated to this issue?

closed 2018-11-18 by HotelCalifornia