purduesigbots/pros-cli #8: Conductor upgrade produces the wrong error message

Octogonapus, 2018-09-12:
#### Expected Behavior:

When I run `prosv5 c u` in a project where everything is up to date, I expect to see something like
```
kernel@3.1.0 is up-to-date.
okapilib@3.2.0 is up-to-date.
```

#### Actual Behavior:

At the least, the current error is wrong. `Upgrading is allowed` isn't allowed if it gives me an error! :)

```
Upgrading kernel
WARNING - pros.conductor.conductor:apply_template - Could not install kernel@3.1.0 because it is not new to the project. Upgrading is allowed, and installing is not allowed
Upgrading okapilib
WARNING - pros.conductor.conductor:apply_template - Could not install okapilib@3.2.0 because it is not new to the project. Upgrading is allowed, and installing is not allowed
```

#### Steps to reproduce:

Run `prosv5 c u` in a project that's up to date.

#### System information:
Operating System:  (Windows, OS X, Ubuntu)

Ubunty 18.04

PROS Version: (run `pros --version`)

3.1.0

#### Additional Information

#### Screenshots/Output Dumps/Stack Traces

HotelCalifornia, 2019-01-30:
closed in #38

HotelCalifornia, 2019-01-30:
not closed yet, oops

edjubuh, 2019-01-30:
Error message now looks like:
```
pros.conductor.project.template_resolution.InvalidTemplateException: Could not install kernel@3.1.6 because it is AlreadyInstalled, and that is not allowed.
```

closed 2019-01-30 by edjubuh

Octogonapus, 2019-01-30:
I would much rather get a message like, "kernel@3.1.6 is already installed.". Exceptions are scary.

PR #38 (2019-01-30): ✨✨🚸 Implement Interactive UI, migrate to sentry-sdk
Summary:
- Implement interactive UI. Interactive UI is a way for the CLI to present the UI to the users and have a (possibly) external presenter worry only about presentation of the CLI's requested elements and alerting the CLI to changes to those elements. The CLI then is responsible for deciding what UI elements are needed and the relevant business logic
- Migrate to new sentry-sdk which works better ™

Closes #17, closes #8

Test Plan:
Dropdown boxes and dialog boxes don't seem to show up on the screen recorder. Not sure why, but they're t [... cut]