purduesigbots/pros-cli #9: Cannot upload binaries to V5 brain on Mac

Qrtn, 2018-09-22:
#### Expected Behavior:

Successful upload of binary to microcontroller.

#### Actual Behavior:

```
Uploading /Users/jeffrey/Documents/hello/bin/output.bin to v5 device on /dev/cu.usbmodem1421 as hello to slot 1
Uploading slot_1.bin (/Users/jeffrey/Documents/hello/bin/output.bin) [#-----------------------------------]   2%
ERROR - pros.cli.upload:upload - Couldn't find the response header in the device response. Got  but was expecting aa55
  File "/usr/local/Cellar/pros-cli/3.1.2/libexec/lib/python3.7/site-packages/pros/serial/devices/vex/vex_device.py", line 69, in _rx_packet
    raise IOError(f"Couldn't find the response header in the device response. "
OSError: Couldn't find the response header in the device response. Got  but was expecting aa55
```

#### Steps to reproduce:

1. `prosv5 conductor new-project test-upload-mac`
2. `cd test-upload-mac`
3. `prosv5 make`
4. `prosv5 upload`

#### System information:
Operating System: macOS High Sierra (10.13.6)

PROS Version: 3.1.2

#### Additional Information

I installed PROS 3 using the instructions for macOS installs involving Homebrew for the latest release (3.1.2). Uploading works fine on another computer running Windows 10.

The percentage at which `prosv5 upload` fails and outputs the above error message varies from about 0 to 25.

#### Screenshots/Output Dumps/Stack Traces

[Full output of `prosv5 --debug upload`](https://github.com/purduesigbots/pros-cli3/files/2407456/prosv5.debug.upload.txt)

jwalk511, 2018-09-22:
We're trying to reproduce; in the meantime could you try restarting your Mac and the V5, and if that fails, plugging the V5 into a different port on the Mac? We'll respond in a bit after more testing.

Qrtn, 2018-09-22:
Will do, thanks for the immediate response!

Qrtn, 2018-09-22:
I've restarted my Mac and the V5, which seemed to have no effect. Using different USB ports, however, yielded varying degrees of success.

- Uploading with either of the two onboard USB ports on my Macbook Pro has never succeeded, with percentages varying from 0 to 40.
- When I connected a powered USB 3.0 hub to my Mac and the V5 to the hub, uploading has succeeded (100%) approximately 1 in 2 times, albeit very slow (around 1 min 30 sec, whereas on another Windows machine the upload takes below 3 sec). When it did not succeed, percentages were in the 90s and close to completion.

The errors when uploading fails are all identical to my original error.

jwalk511, 2018-09-22:
What kind of USB ports are on your Pro? USB C or A? And what do you get when you run `prosv5 v5 status`?

Qrtn, 2018-09-22:
USB A 3.0. `prosv5 v5 status` for one of the onboard ports:

```
Connected to V5 on /dev/cu.usbmodem1421
System version: 1.0.0-0
CPU0 F/W version: 1.0.0-23
CPU1 SDK version: 1.0.0-0
System ID: 0xd5442800
```

For the hub:

```
Connected to V5 on /dev/cu.usbmodem14141
System version: 1.0.0-0
CPU0 F/W version: 1.0.0-23
CPU1 SDK version: 1.0.0-0
System ID: 0xd5442800
```

jwalk511, 2018-09-22:
Thanks for the info. 

The System version, CPU0 F/W version, and CPU1 SDK version all match what I've been testing with, and we're both using USB 3.0 type A. The only difference I've noticed between us is the size of the upload file, and that's because I've been using an older version of gcc-arm-none-eabi, so I'm not sure why I can't reproduce.

Do you have any other devices you can connect to you computer through USB? Have you noticed that they have or have not been communicating well?

jwalk511, 2018-09-22:
If you suspect that the physical ports may be an issue you can follow the steps at https://www.makeuseof.com/tag/mac-hardware-diagnostic-tests/ to run Apple Diagnostics and see if you get error code `NDD001` (Possible problems with USB hardware), but I doubt that's the issue since you do get some connectivity. We can try giving you a version of the uploader with delays built in so it has more time to receive a response, though that may not come until later.

Qrtn, 2018-09-22:
Thanks very much for your help. I've never really had a problem with other USB devices. I will be testing it on another MacBook (though that will take some time); meanwhile I am also testing uploads on another Windows computer.

Are you also using a Mac? And do your upload speeds take upwards of a minute as well?

jwalk511, 2018-09-22:
I'm using macOS High Sierra version 10.13.6 on a 2017 27-inch iMac through a USB A 3.0 port with prosv5 cli version 3.1.2.

The only appreciable differences I've noticed is that

- I'm not on a laptop

- I'm possibly running a computer made in a different year

My upload speeds in in a few seconds, but your slow upload speeds are almost certainly related to whatever is causing total failure

jwalk511, 2018-09-22:
I'm going to sign off for the night. We'll test with other machines as they become available, but I'm afraid there isn't much we can do until we can reproduce the issue

HotelCalifornia, 2018-09-22:
we pushed a potential fix to the develop branch. you can test it by running the following commands:

```
> brew update
> brew unlink pros-cli
> brew install pros-cli --HEAD
```

Qrtn, 2018-09-22:
Unfortunately the upload didn't work. The ini and bin file upload were switched though.

```
Uploading /Users/jeffrey/Documents/test-upload-mac/bin/output.bin to v5 device on /dev/cu.usbmodem1421 as test-upload-mac to slot 1                                                                      
Uploading slot_1.ini [####################################] 100%
Uploading slot_1.bin (/Users/jeffrey/Documents/test-upload-mac/bin/output.bin) [#-----------------------------------]   2% 0d 00:02:12                                                                   
ERROR - pros.cli.upload:upload - Couldn't find the response header in the device response. Got  but was expecting aa55                                                                                   
  File "/usr/local/Cellar/pros-cli/HEAD-df2850d/libexec/lib/python3.7/site-packages/pros/serial/devices/vex/vex_device.py", line 69, in _rx_packet                                                       
    raise IOError(f"Couldn't find the response header in the device response. "
OSError: Couldn't find the response header in the device response. Got  but was expecting aa55
```

[Full output of prosv5 --debug upload](https://github.com/purduesigbots/pros-cli3/files/2408082/prosv5.debug.upload.2.txt)

edjubuh, 2018-09-22:
Can you run those brew commands again? I've halved the packet size from 4096 byte chunks to 2048 byte chunks to see if that helps resolve your issue.

Qrtn, 2018-09-22:
That didn't seem to do it either. Most of the uploads now actually break at 0 percent.

HotelCalifornia, 2018-09-22:
Just to make sure, you don't have VCS open while you're using PROS to upload, right?

Qrtn, 2018-09-22:
No, VCS is closed. When it is open, PROS says that the resource is busy.

Qrtn, 2018-09-25:
We just tested the PROS uploading on another MacBook Pro on High Sierra 10.13.6. It's having the same error as the one on the original Mac. So I think this is macOS-specific issue.

Qrtn, 2018-10-09:
I have. I can try using a different USB cable tomorrow.

HotelCalifornia, 2018-10-21:
@Qrtn did you ever try a new cable?

NoMod-Programming, 2018-11-04:
Hi, I'm hoping that I can provide a bit of info here, since I've had a similar issue with Ubuntu 18.04 on a Macbook Pro (2013), and found a bit of a workaround: If I plug in a v5 brain, wait about a minute or two without trying to access any of the serial ports for the brain, and then download, it works. This is with multiple USB cables, and `fuser /dev/ttyACM0` as root doesn't show any processes using the serial port until then, even though it's "in use" for about a minute until I can actually open it.

HotelCalifornia, 2018-12-19:
Closing due to inactivity. If you're still having problems with this, feel free to reopen

pieceofpie, 2018-12-22:
I have the same error to upload to v5 brain with macbook. window seems fine

HotelCalifornia, 2018-12-22:
We're still looking for a reliable way to reproduce this issue. Any further information you can provide regarding what was happening when you got this error will be helpful

pieceofpie, 2018-12-22:
I install pros as suggest in the instruction
install brew
brew tap purduesigbots/pros
brew cask install gcc-arm-embedded pros-editor.

I also install atom git python separately.
I also try to uninstall and reinstall several time. Same error when upload to v5 brain.

Let me know anything else you need to know.

pbchase, 2018-12-31:
I am also unable to upload from MacOSX to V5. I am running pros-cli 3.1.3 on MacOSX High Sierra 10.13.6:

```
→ brew list pros-cli
/usr/local/Cellar/pros-cli/3.1.3/bin/prosv5
/usr/local/Cellar/pros-cli/3.1.3/libexec/bin/ (27 files)
/usr/local/Cellar/pros-cli/3.1.3/libexec/lib/ (2378 files)
/usr/local/Cellar/pros-cli/3.1.3/libexec/.Python
```

Here's a log of my build and upload session
```
(20:50:29) ± [pbc@MyMac.local] ~/vex_projects/okapi_drive (master ✗) 
→ prosv5 build all
Cleaning project
-n Compiling src/autonomous.cpp 
-e \x1b[32;01m[OK]\x1b[0m
-n Compiling src/initialize.cpp 
-e \x1b[32;01m[OK]\x1b[0m
-n Compiling src/opcontrol.cpp 
-e \x1b[32;01m[OK]\x1b[0m
-n Adding timestamp 
-e \x1b[32;01m[OK]\x1b[0m
-n Linking project with libpros,okapilib 
-e \x1b[32;01m[OK]\x1b[0m
Section sizes:
   text	   data	    bss	  total	    hex	filename
459.35KB  3.22KB  2.59MB  3.04MB  3098a6 bin/output.elf
-n Creating bin/output.bin for VEX EDR V5 
-e \x1b[32;01m[DONE]\x1b[0m
Capturing metadata for PROS Editor...
(20:51:04) ± [pbc@MyMac.local] ~/vex_projects/okapi_drive (master ✗) 
→ prosv5 lsusb
VEX EDR V5 System Ports:
/dev/cu.usbmodem1411 - VEX Robotics V5 Brain - 00000000
VEX EDR V5 User ports:
/dev/cu.usbmodem1413 - VEX Robotics V5 Brain - 00000000
There are no connected VEX EDR Cortex Microcontroller Ports
(20:51:10) ± [pbc@MyMac.local] ~/vex_projects/okapi_drive (master ✗) 
→ prosv5 upload --slot=3
Uploading /Users/pbc/vex_projects/okapi_drive/bin/output.bin to v5 device on /dev/cu.usbmodem1411 as okapi_drive to slot 3
Uploading slot_3.ini [####################################] 100%
Uploading slot_3.bin (/Users/pbc/vex_projects/okapi_drive/bin/output.bin) [##----------------------------------]   6% 0d 00:00:55
ERROR - pros.cli.upload:upload - Couldn't find the response header in the device response. Got  but was expecting aa55
  File "/usr/local/Cellar/pros-cli/3.1.3/libexec/lib/python3.7/site-packages/pros/serial/devices/vex/vex_device.py", line 69, in _rx_packet
    raise IOError(f"Couldn't find the response header in the device response. "
OSError: Couldn't find the response header in the device response. Got  but was expecting aa55
We detected something went wrong! Would you like to send a bug report to the PROS Development team? [Y/n]: y
Sending report...
Sentry is attempting to send 1 pending error messages
Waiting up to 10 seconds
Press Ctrl-C to quit
(20:51:39) ± [pbc@MyMac.local] ~/vex_projects/okapi_drive (master ✗) 
```

I ran the brew commands to reinstall the pros-cli from HEAD, but the results were differently bad:

```
→ brew list pros-cli
/usr/local/Cellar/pros-cli/HEAD-96ce7d0/bin/prosv5
/usr/local/Cellar/pros-cli/HEAD-96ce7d0/libexec/bin/ (27 files)
/usr/local/Cellar/pros-cli/HEAD-96ce7d0/libexec/lib/ (2374 files)
/usr/local/Cellar/pros-cli/HEAD-96ce7d0/libexec/.Python
(21:46:16) ± [pbc@MyMac.local] ~/vex_projects/okapi_drive (master ✗) 
→ prosv5 lsusb
Traceback (most recent call last):
  File "/usr/local/Cellar/pros-cli/HEAD-96ce7d0/libexec/bin/prosv5", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/local/Cellar/pros-cli/HEAD-96ce7d0/libexec/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3126, in <module>
    @_call_aside
  File "/usr/local/Cellar/pros-cli/HEAD-96ce7d0/libexec/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3110, in _call_aside
    f(*args, **kwargs)
  File "/usr/local/Cellar/pros-cli/HEAD-96ce7d0/libexec/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3139, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/local/Cellar/pros-cli/HEAD-96ce7d0/libexec/lib/python3.7/site-packages/pkg_resources/__init__.py", line 581, in _build_master
    ws.require(__requires__)
  File "/usr/local/Cellar/pros-cli/HEAD-96ce7d0/libexec/lib/python3.7/site-packages/pkg_resources/__init__.py", line 898, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/local/Cellar/pros-cli/HEAD-96ce7d0/libexec/lib/python3.7/site-packages/pkg_resources/__init__.py", line 784, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'rfc6266' distribution was not found and is required by pros-cli-v5
(21:46:19) ± [pbc@MyMac.local] ~/vex_projects/okapi_drive (master ✗) 
```

HotelCalifornia, 2018-12-31:
Yeah, don't install from HEAD. The last thing we pushed there was a test, and it hasn't been updated since.

Someone on discord the other day reminded me of #29, which may fix this issue. I'll update the branch the HEAD formula points to later today with that fix, so we can see if that works.

pbchase, 2018-12-31:
Yeah, I backed the HEAD install out pretty fast. ;-)  Let me know when you need a test of the PR #29 commits.

HotelCalifornia, 2019-01-01:
@pbchase I've updated and pushed the test change to HEAD. let me know how it goes

pieceofpie, 2019-01-01:
recently update to the MacOS mojave and reinstall the pros. stuck in the uploading forever.

brew cask reinstall gcc-arm-embedded pros-editor

prosv5 upload --slot 5
MacBookPro:v5debug $ prosv5 upload --slot 5
Uploading /Users/BitTorrent Sync/v5debug/bin/output.bin to v5 device on /dev/cu.usbmodem14101 as v5debug to slot 5
Uploading slot_5.ini [####################################] 100%
Uploading slot_5.bin (/Users/BitTorrent Sync/v5debug/bin/output.bin) [------------------------------------]   0%

pbchase, 2019-01-04:
I reinstalled pros-cli with the --HEAD option, but it failed again in the same way as the initial report:

```
ERROR - pros.cli.upload:upload - Couldn't find the response header in the device response. Got  but was expecting aa55
  File "/usr/local/Cellar/pros-cli/HEAD-ae037ea/libexec/lib/python3.7/site-packages/pros/serial/devices/vex/vex_device.py", line 69, in _rx_packet
    raise IOError(f"Couldn't find the response header in the device response. "
OSError: Couldn't find the response header in the device response. Got  but was expecting aa55
```

Here's the entire session:

```
(20:43:00) ± [pbc@MyMac.local] ~/vex_projects/okapi_drive (master ✓) 
→ brew list pros-cli
/usr/local/Cellar/pros-cli/HEAD-ae037ea/bin/prosv5
/usr/local/Cellar/pros-cli/HEAD-ae037ea/libexec/bin/ (27 files)
/usr/local/Cellar/pros-cli/HEAD-ae037ea/libexec/lib/ (2378 files)
/usr/local/Cellar/pros-cli/HEAD-ae037ea/libexec/.Python
(20:43:16) ± [pbc@MyMac.local] ~/vex_projects/okapi_drive (master ✓) 
→ prosv5 build all
Cleaning project
-n Compiling src/autonomous.cpp 
-e \x1b[32;01m[OK]\x1b[0m
-n Compiling src/initialize.cpp 
-e \x1b[32;01m[OK]\x1b[0m
-n Compiling src/opcontrol.cpp 
-e \x1b[32;01m[OK]\x1b[0m
-n Adding timestamp 
-e \x1b[32;01m[OK]\x1b[0m
-n Linking project with libpros,okapilib 
-e \x1b[32;01m[OK]\x1b[0m
Section sizes:
   text	   data	    bss	  total	    hex	filename
459.44KB  3.22KB  2.59MB  3.04MB  30990e bin/output.elf
-n Creating bin/output.bin for VEX EDR V5 
-e \x1b[32;01m[DONE]\x1b[0m
Capturing metadata for PROS Editor...
(20:45:08) ± [pbc@MyMac.local] ~/vex_projects/okapi_drive (master ✓) 
→ prosv5 lsusb
VEX EDR V5 System Ports:
/dev/cu.usbmodem1421 - VEX Robotics V5 Brain - 00000000
VEX EDR V5 User ports:
/dev/cu.usbmodem1423 - VEX Robotics V5 Brain - 00000000
There are no connected VEX EDR Cortex Microcontroller Ports
(20:45:13) ± [pbc@MyMac.local] ~/vex_projects/okapi_drive (master ✓) 
→ prosv5 upload --slot=3
Uploading /Users/pbc/vex_projects/okapi_drive/bin/output.bin to v5 device on /dev/cu.usbmodem1421 as okapi_drive to slot 3
Uploading slot_3.ini [####################################] 100%
Uploading slot_3.bin (/Users/pbc/vex_projects/okapi_drive/bin/output.bin) [------------------------------------]   2% 0d 00:05:26
ERROR - pros.cli.upload:upload - Couldn't find the response header in the device response. Got  but was expecting aa55
  File "/usr/local/Cellar/pros-cli/HEAD-ae037ea/libexec/lib/python3.7/site-packages/pros/serial/devices/vex/vex_device.py", line 69, in _rx_packet
    raise IOError(f"Couldn't find the response header in the device response. "
OSError: Couldn't find the response header in the device response. Got  but was expecting aa55
We detected something went wrong! Would you like to send a bug report to the PROS Development team? [Y/n]: y
Sending report...
Sentry is attempting to send 1 pending error messages
Waiting up to 10 seconds
Press Ctrl-C to quit
(20:45:35) ± [pbc@MyMac.local] ~/vex_projects/okapi_drive (master ✓) 
```

HotelCalifornia, 2019-01-04:
Thanks for testing. Can you try uploading again with the `--debug` flag and share the output of that?

oliverwjay, 2019-01-04:
I am having the same issue on High Sierra with PROS cli 3.1.3 installed using pip. I am getting the following output from `prosv5 upload --debug`
[PROS Upload response.txt](https://github.com/purduesigbots/pros-cli3/files/2727566/PROS.Upload.response.txt)

Bottersnike, 2019-01-04:
@oliverwjay If you go into the installed pros source, then go to `pros/serial/devices/vex/vex_device` and change the `0.01` in the line
```py
def _rx_packet(self, timeout: float = 0.01) -> Dict[str, Union[Union[int, bytes, bytearray], Any]]:```
(around line 50) to something like `timeout: float = 1.0` do you still get the same error?

oliverwjay, 2019-01-04:
It appears that it is not exiting because of the timeout. Even setting the timeout to 30s, it is still raising the error within a few seconds.

Bottersnike, 2019-01-04:
Are you able to toss `print(repr(b))` just after `b = self.port.read(1)` to see what's going on there?

oliverwjay, 2019-01-04:
Here is the output
[PROS Upload rep b.txt](https://github.com/purduesigbots/pros-cli3/files/2727825/PROS.Upload.rep.b.txt)

Bottersnike, 2019-01-04:
And does the error persist if you replace `time.time() - start_time < timeout` with `True`?

oliverwjay, 2019-01-04:
It stops responding. It appears to be receiving a timeout of .1 _txrx_ext_packet in v5_device.py. By increasing that it lasts longer, but still errors.

oliverwjay, 2019-01-04:
Switching between my usb ports has not had any noticeable affect for me

HotelCalifornia, 2019-01-04:
> Switching between my usb ports has not had any noticeable affect for me

Okay, that's about what I expected.

oliverwjay, 2019-01-04:
I just tried running it through a simple USB 2.0 hub and it worked fine. I am guessing it has to do with the unusual handling of the USB 3.0/2.0 combined ports

Bottersnike, 2019-01-06:
I have achieved highly reproducible results:
```diff
+ [OK ] Laptop USB 3
+ [OK ] Laptop USB 2

- [ERR] Dell FPIO "USB2 Enhanced Host controller"
- [ERR] Dell FPIO "USB 3.0 eXtensible Host Controller"
+ [OK ] Dell FPIO "USB2 Enhanced Host controller" -> USB 2.0 Hub

+ [OK ] Dell RPIO "USB2 Enhanced Host controller"
+ [OK ] Dell RPIO "USB 3.0 eXtensible Host Controller"
+ [OK ] Dell RPIO "USB2 Enhanced Host controller" -> USB 2.0 Hub
```

I'm thinking the problem is due to a difference in the physical connections to front-panel I/O (in this case) rather than rear-panel I/O. Without a doubt though, the workaround is to use a USB Hub.

Bottersnike, 2019-01-06:
Update 8 minutes later: Just used the firmware update utility and it communicated fine in all situations. Then went back and tested pros: Worked perfectly.

HotelCalifornia, 2019-01-06:
Note also that there seem to be two problems:
1. The edge case in packet size detection which is fixed by #29
2. The weird USB behavior as documented in @Bottersnike's comments

pbchase, 2019-01-12:
I'm also having good results with the USB 2.0 hub workaround. Downloads complete in under 5 seconds.

HotelCalifornia, 2019-01-16:
Made the title a little better than it was before

Heasummn, 2019-01-26:
I am also experiencing this issue on a Windows laptop. I have tried switching brains, ports and USB cables. Currently, I am reinstalling PROS. The exact error says: 
```Uploading slot_1.bin (D:\files\vex\turning_point\bin/output.bin) [###################################-]  99% 0d 00:00:01
ERROR - pros.cli.upload:upload - Couldn't find the response header in the device response. Got  but was expecting aa55
  File "C:\Users\PROS\Jenkins\workspace\master95eda9d1\pros\serial\devices\vex\vex_device.py", line 109, in _txrx_packet
  File "C:\Users\PROS\Jenkins\workspace\master95eda9d1\pros\serial\devices\vex\vex_device.py", line 69, in _rx_packet
OSError: Couldn't find the response header in the device response. Got  but was expecting aa55
ERROR - __main__:main - name 'exit' is not defined
  File "C:\Program Files\Python36\lib\site-packages\click\core.py", line 535, in invoke
  File "C:\Users\PROS\Jenkins\workspace\master95eda9d1\pros\cli\upload.py", line 118, in upload
```

The directory `C:/Users/PROS` does not exist, but I doubt that is related to the problem. My download halts at 99%, but the download does complete and install the updated code onto the robot. This problem happened randomly after PROS working consistently for 2 months. I will report back if reinstalling PROS solves the problem.

Bottersnike, 2019-01-26:
If reinstalling PROS does not fix the problem can you also try (in this order):
- Using an external USB hub
- Regardless if ^ worked, opening the firmware updater, closing it without updating, then retrying

DoctorFogarty, 2019-01-27:
Hello,

I just started having this problem this evening on a Windows PC. I was building and deploying code successfully for 1-2 hours before the problem appeared. I was running a project updated to the Latest API and OKAPI.

Swapping USB, rebooting the PC/V5 system, and just restarting PROS didn't help. I event copied the entire project onto a flash drive and tried building and deploying the project on a different laptop and got the same error. 

To continue doing any work I ended up doing was opening up a project that was still on the 3.1.1 PROS API version.... and copied all of my SRC files into the project. This project running the old API version built and deployed successfully. 

Not sure what other information I could give that might help. It's very strange that everything worked fine and then something exploded and now nothing seems to work on the latest API version. I have not tried re-installing PROS yet.

Heasummn, 2019-01-27:
I actually solved the issue, although the cause was rather silly. It seems like the firmware was corrupted, making a new project and copying the source code over solved all issues.

DoctorFogarty, 2019-01-27:
> I actually solved the issue, although the cause was rather silly. It seems like the firmware was corrupted, making a new project and copying the source code over solved all issues.

That's odd....why would anything we would do in the build/deploy process do anything to the firmware?

HotelCalifornia, 2019-01-27:
> I'm unsure, I tried isolating every variable, and eventually I realized the only thing I hadn't tried was the code. My code is in a USB drive and I fear that the firmware may have gotten corrupted. On that note, does pros have some sort of `clean` function? A function that re-downloads the firmware and removes the entire bin folder.

what firmware are you talking about? we have a clean function (by virtue of GNU Make) that will remove the contents of the bin folder (running `prosv5 make clean` does this alone, and running `prosv5 make all` will clean and then build).

Heasummn, 2019-01-28:
I meant to say the PROS library itself. The files contained in /firmware were the ones that somehow got corrupted and were preventing the download. This was confirmed by making a new project and copying the folder over to the failing project, which solved the problem.

DoctorFogarty, 2019-01-29:
I'm not sure how a USB hub would help in my case. 
I was using an older labtop with USB 2.0 ports when I saw this issue.

I created a new project on the latest API and haven't had the problem appear again.

ThadHouse, 2019-02-03:
I'm noticing this issue seems to be caused by how big a file is. I was having this issue, and both adding a line of code and removing a line of code fixed the issue. I wonder if there is something in the upload utilities having issues with certain file sizes.
 
If it helps, I'm attaching 3 binaries. The failure is the failure, the one with no extension is removing a line, the one with addedline is adding a line. Also attached is the project we're seeing the issue with. 

[vexprosuploadfailures.zip](https://github.com/purduesigbots/pros-cli/files/2826081/vexprosuploadfailures.zip)

[CompPros.zip](https://github.com/purduesigbots/pros-cli/files/2826083/CompPros.zip)

HotelCalifornia, 2019-02-03:
@ThadHouse that is the first issue I mention in this comment: https://github.com/purduesigbots/pros-cli/issues/9#issuecomment-451751459

A patch was merged in (thanks @Bottersnike) and will be available with the next release of the CLI.

oliverwjay, 2019-02-18:
> I just wanted to re-emphasize that I've been successfully uploading binaries on an Arch Linux VM on my MacBook Pro for a while now (uploading on Mac itself still doesn't work). So maybe it is an OS-specific issue. I'm really not sure if this has ever happened so far on other Windows laptops (with USB 3.0 ports) and desktops that I have tested.

I would like to add that I have also had no issues downloading when I boot into Ubuntu on the same computer that is experiencing issues in macOS.

HotelCalifornia, 2019-02-18:
@Qrtn / @oliverwjay did either of you try using a USB hub when uploading?

oliverwjay, 2019-02-18:
> @Qrtn / @oliverwjay did either of you try using a USB hub when uploading?

From macOS I have never had success without a USB hub, and it works some days (I have not identified a pattern) using an unpowered USB3.0 hub. 

From Ubuntu, it consistently works with or without the hub.

HotelCalifornia, 2019-02-18:
> (I have not identified a pattern)

I wonder if it's what @Bottersnike suggested here https://github.com/purduesigbots/pros-cli/issues/9#issuecomment-457858981 regarding using the firmware utility (or VCS) to configure the port first

[3 more replies]

closed 2022-01-18 by kunwarsahni01