purduesigbots/pros-cli #10: File not found error when invoking prosv5 make

gnations56, 2018-09-29:
#### Expected Behavior:
The project will build
#### Actual Behavior:
make fails with file not found error. PROS is not specific as to which file is missing
#### Steps to reproduce:
run prosv5 make in a project folder
#### System information:
Operating System:  (Windows, OS X, Ubuntu)
Windows 10
PROS Version: (run `pros --version`)

#### Additional Information

#### Screenshots/Output Dumps/Stack Traces
ERROR - __main__:main - [WinError 3] The system cannot find the path specified
  File "C:\Program Files\Python36\lib\subprocess.py", line 709, in __init__
  File "C:\Program Files\Python36\lib\subprocess.py", line 997, in _execute_child
FileNotFoundError: [WinError 3] The system cannot find the path specified

EricHallahan, 2018-11-05:
Sorry to bump the thread. I have the identical issue after installing PROS manually (i.e. Installing the GNU Toolchain and wheel). What was the solution? I am afraid that the provided solution is not entirely obvious.

HotelCalifornia, 2018-11-06:
@EricHallahan what platform are you on? the OP's issue was related to using the incorrect Windows installer for their system.

If you're also on Windows, make sure that you've set the `PROS_TOOLCHAIN` environment variable.

EricHallahan, 2018-11-07:
Windows 10 1809, 64-bit. What is in the `PROS_TOOLCHAIN` variable? I didn't use the installer. I assumed that adding GNU Tools to `PATH` would be enough.

jonathanloh02, 2019-01-24:
I have a similar problem as well

#### Error
`prosv5 make`:
```
ERROR - __main__:main - [WinError 2] The system cannot find the file specified 
File "C:\Program Files\Python36\lib\subprocess.py", line 997, in _execute_child
# ReadFile will hang.
```

#### System information:
Operating System: Windows 10 64-bit operating system, x64-based processor

PROS Version: (run `pros --version`) pros, version 3.1.3

#### Additional Information
Installer used: pros-windows-3.1.3.0-64bit
Building from GUI does not seem to work - changes to program are not reflected once uploaded to robot

#### Screenshots/Output Dumps/Stack Traces
`echo %PATH%` (Shortened):
```
C:\Program Files\PROS\cli;
C:\Program Files\Python36\Scripts\;
C:\Program Files\Python36\;
C:\Users\Jonathan\.windows-build-tools\python27\;
C:\Program Files\PROS\cli;
C:\Program Files\PROS\toolchain\bin;
C:\Program Files\Python36\lib;
C:\Program Files\Git;
```

`echo %PROS_TOOLCHAIN%`:
C:\Program Files\PROS\toolchain\usr

edjubuh, 2019-01-24:
The file subprocess would be failing to execute would be make.exe. It should exist in %PROS_TOOLCHAIN%\bin, but can you verify this?

jonathanloh02, 2019-01-24:
Solved the issue. the toolchain in %PATH% should have been `C:\Program Files\PROS\toolchain\usr\bin` instead of `C:\Program Files\PROS\toolchain\bin`

EricHallahan, 2019-01-25:
I forgot to mention I figured my problem out.

HotelCalifornia, 2019-01-25:
@erichallahan care to share? :smile:

3038922, 2019-02-01:
How do you configure vscode?

EricHallahan, 2019-02-11:
> And how do you configure vscode?

By running `prosv5 mu` in tasks.
```
{
    // See https://go.microsoft.com/fwlink/?LinkId=733558
    // for the documentation about the tasks.json format
    "version": "2.0.0",
    "tasks": [
        {
            "label": "make-upload",
            "type": "shell",
            "command": "prosv5 mu",
            "group": {
                "kind": "build",
                "isDefault": true
            }
        }
    ]
}
```
I recently thought of using Fortran for linear algebra, but i would need to build my own toolchain for that.

HotelCalifornia, 2019-02-11:
> I recently thought of using Fortran for linear algebra, but I would need to build my own toolchain for that.

you'd need to do a little more than just that... as far as I can tell, the only supported Fortran compiler for arm requires a linux environment

EricHallahan, 2019-02-11:
I will likely install Linux soon on my device, either as a VM or with wubi.

HotelCalifornia, 2019-02-11:
No, that's not what I meant. I meant that the only compiler for fortran on arm is gfortran-arm-linux, which requires the target system have a linux environment. in this case, that would be the v5