purduesigbots/pros-cli #11: PROS will not create project on 32-bit system

RyzenFromFire, 2018-10-02:
#### Expected Behavior:
PROS should create a working and editable project to be used with V5.

#### Actual Behavior:
PROS does nothing once the "Create Project" button is clicked aside from exiting the dialog.

#### Steps to reproduce:
Install 32-bit PROS on a 32-bit system (win7).
(steps I tried to fix:) Restart PROS several times. Install Java and manually replace the toolchain with the win32 one from the latest release. 
Attempt to create a project with the "New Project" button from the PROS menu.
Name a valid directory and select everything properly.
Press the "Create Project" button.

#### System information:
Operating System:  Windows 7 32-bit (pretty sure it's home edition).
Processor i3 @ 2.4GHz
4GB RAM (2.88 usable since it's 32-bit)
The PC is a HP ProBook 4000 series.

PROS Version: (run `pros --version`) 3.1.2 (latest)

#### Additional Information
I have a 64-bit PC with 64-bit PROS installed, and it works fine.

#### Screenshots/Output Dumps/Stack Traces
There was one error that I got when attempting to run the "pros conduct first-run" command through terminal, I couldn't run any commands through the terminal either.

`Error starting process:
Error: Cannot find module 'C:\Users\User\.pros-editor\packages\atom-ide-ui\node_modules\nuclide-prebuilt-libs\pty\build\pty-electron-v2.0-win32-ia32\pty.node'`

HotelCalifornia, 2018-11-18:
I'm not sure why the `prosv5` command isn't getting registered. are you still having this issue with the latest version of the PROS installer?

RyzenFromFire, 2018-11-28:
Will test tomorrow.

RyzenFromFire, 2018-11-28:
Just tested it, the new version makes no difference. Upon creating project, the dialog closes and nothing happens. No folder is created at the selected path. 

When I open up the terminal from the PROS toolbar, I get the same message I got before: 
`Error starting process: Error: Cannot find module 'C:\Users\User\.pros-editor\packages\atom-ide-ui\node_modules\nuclide-prebuilt-libs\pty\build\pty-electron-v2.0-win32-ia32\pty.node'`

HotelCalifornia, 2018-12-19:
The terminal issue is separate- it's notoriously buggy. in the meantime, the solution is to use an external terminal.

Did you check to make sure whether the `prosv5` command was installed after the last time?

HotelCalifornia, 2019-01-01:
> How would I check if the command is installed?

running `prosv5 --version` (or any other CLI command) from the command line