purduesigbots/pros-cli #14: Request added functionality: update only required files

3038922, 2018-10-14:
In order to facilitate some classmates like my English is not good, I put the API comments translated into Chinese. But when I use "prosv5 conduct apply prosv5" or "okapilib", he will I just translated the annotation ". HPP "file all the covers. I hope I can get a command, his role is only for the update". HPP "file to replace. Thank you

edjubuh, 2018-10-14:
That's not something we're currently interested in supporting since we'd like to do our best to guarantee that all users have the correct versions of header and code files. 

There are some additional challenges with determining when we'd want to replace header files - we can compare the hashes of the file that was originally installed, the current file, and the new header file. Under your system, you'd want to compare the original file hash with the new header file hash and only replace when that changes. It makes more sense from my perspective as the library maintainer to compare the current file hash with the new file hash - I'm interested in making sure you have the latest files.

There is an alternative solution which is a bit farther out on the horizon, which is to use template metadata to differentiate locales for these comments. A workaround that's closer to reality could be to maintain a fork of the PROS kernel with patched Chinese docstrings; then we can help maintain a depot with all translations of all the templates.

3038922, 2018-10-14:
I am too selfish to take your workload into consideration. Thank you very much and your team for all that you have done for me.

HotelCalifornia, 2019-01-04:
I'm going to close this for now. if we end up getting around to working on the solutions @edjubuh mentioned, I'll reopen

closed 2019-01-04 by HotelCalifornia