purduesigbots/pros-cli #15: Linter not working and cquery not initializing

py2025, 2018-10-16:
#### Expected Behavior: 
Errors should be detected under diagnostics and should be underlined in the code.

#### Actual Behavior:
No errors are marked even if done purposely, and programs will not build.

#### Steps to reproduce:
Downloaded PROS like this

#### System information:
Operating System:  (Windows, OS X, Ubuntu)

OS X 10.13.2

PROS Version: (run `pros --version`)
PROS Version 3.1.0
#### Additional Information

#### Screenshots/Output Dumps/Stack Traces

edjubuh, 2018-10-16:
Can you share some additional information? Can share the output of running build?

HotelCalifornia, 2018-10-17:
This is a problem with the CLI on macOS where the intercept-cc and intercept-cc++ scripts are installed to a location not on PATH. There is a fix for this staged for the next release.

@Rybread30 the problem you're having where cquery can't be found seems to be a separate issue. How did you install the PROS Editor?

HotelCalifornia, 2018-10-30:
following the steps detailed [here](https://github.com/purduesigbots/pros-cli3/releases/3.1.2)?

Rybread30, 2018-10-30:
Yes.

HotelCalifornia, 2018-11-18:
we're planning to have a release out soon that will fix this issue with cquery binaries not being on PATH. sorry for the inconvenience.

dongwon900, 2018-11-21:
the code does upload fine when I use make while in the project directory. I don't know enough to understand why this is, but for now I'm willing to deal with the minor inconvenience of doing this for now.

HotelCalifornia, 2018-11-21:
> the code does upload fine when I use make while in the project directory.

Yeah, the problem is that when you build the project in the editor, we run an extra command to capture some metadata that the linter uses. Unfortunately, after migrating to the homebrew formula, two of the required binaries for this process are not accessible by the CLI. The fix (which should be out in the near future) works around this issue.

HotelCalifornia, 2018-11-26:
@Rybread30 @dongwon900 can you verify that these issues are still occurring on the latest version of the PROS CLI, version 3.1.3?

HotelCalifornia, 2018-12-19:
I'm closing this due to inactivity. If any of you are still running into this issue, feel free to reopen

closed 2018-12-19 by HotelCalifornia