purduesigbots/pros-cli #16: maybe "pros::ADIGyro gyro" has some bug

3038922, 2018-10-25:
#### Expected Behavior:
get gyro value
#### Actual Behavior:
it aways return 127.0
#### Steps to reproduce:

#### System information:
Operating System:  win10 1803

PROS Version: (run `pros --version`)
pros, version 3.1.2

#### Additional Information
 in prosv5 3.1.1 ,and i use okapilib it was all right.

update prosv5 to 3.1.2 
okapilib to 3.3.5 
it was wrong
#### Screenshots/Output Dumps/Stack Traces
[image]
[image]

[image]

HotelCalifornia, 2018-10-25:
Thanks for the report. There was in fact an issue with all the legacy sensors, which has been fixed with the release of PROS kernel 3.1.3. you can run `prosv5 conduct upgrade` in the root of your project directory to get this fix.

closed 2018-10-25 by HotelCalifornia

3038922, 2018-10-26:
I have updated the kernel to 3.1.3, but the master return version is still 3.1.1. Then the adi for gyro return value is still 127.
[image]

[image]
[image]

edjubuh, 2018-10-26:
See response [here](https://www.vexforum.com/index.php/35269-prosv5-the-problem-of-setting-the-gyroscope-to-a-global-class/p1#p285857).