purduesigbots/pros-cli #18: Filesystem  maybe have some bug.

3038922, 2018-11-13:
#### Expected Behavior:
Using STL to read and write files
#### Actual Behavior:
1 if i use "std::fstream" ,file can't open
2 If I use C language, I can open and read files. But if there are Chinese characters, PYTHON3 will make mistakes. Maybe it's the encoding of files.
#### Steps to reproduce:

#### System information:
Operating System:  (Windows, OS X, Ubuntu)
windows 10 1803
PROS Version: (run `pros --version`)
pros, version 3.1.2
#### Additional Information
kernel 3.3.4
oklib 3.3.7
#### Screenshots/Output Dumps/Stack Traces
[image]
[image]

3038922, 2018-11-15:
My current solution plan is:
[image]
But the class coming out of NEW will run a little slower. I don't know how much it will affect.

Bottersnike, 2018-11-23:
Would switching the serial terminal encoding to latin-1 (ext ASCII) be a plausible fix as then any byte received will have a character in the 0-255 range to map to?

HotelCalifornia, 2018-11-24:
> Would switching the serial terminal encoding to latin-1 (ext ASCII) be a plausible fix as then any byte received will have a character in the 0-255 range to map to?

We have ASCII hardcoded as the decoding codec in some places, which I think is the problem (as the terminal uses whatever codec is passed down to it).

@3038922 it's possible that the name of your file is too long. Can you try it with a file that has an 8 or fewer character name?

3038922, 2018-11-24:
Okay, I'll try it tomorrow, but ASCII doesn't support Chinese.

HotelCalifornia, 2018-11-24:
> but ASCII doesn't support Chinese.
yes I know, that's why it's a problem.

let me know whether changing the file name length does anything.

3038922, 2018-11-26:
[image]
use c mode is ok!
use c++ mode also error;

HotelCalifornia, 2018-12-20:
from https://github.com/purduesigbots/pros-cli3/issues/33#issue-393091593:
> I've been working with some code which uses `std::ifstream` to manipulate files, and it can open the files, but cannot read or write. `tellg` seems to always return -1. I can convert it to use `fopen` and `fread` and it works perfectly. I've had to do this for multiple projects, so it seems like a problem with PROS.

We've observed the same behavior, but further investigation was inconclusive as to why `tellg` would behave in such a way. Perhaps @edjubuh can weigh in