purduesigbots/pros-cli #19: User Defined Tasks end between Autonomous and Driver modes

onthegrid007, 2018-11-17:
#### Expected Behavior:
User Defined Tasks function in-between Autonomous and Driver controlled modes

#### Actual Behavior:
User Defined Tasks end between Autonomous and Driver controlled modes

#### Steps to reproduce:
1. Make basic User Defined Display Task that displays the button presses
2. Plug in "VEXnet COMPETITION SWITCH" to v5 remote and set to 'Enabled' and 'Driver'
3. Switch to 'Disabled' on the Competition Switch, and the User defined task will stops functioning

#### System information:
Operating System:  Windows 10

pros, version 3.1.2

#### Additional Information
In Robotc we, Team 987, used to program with in the past years contained a boolean value in the competition template that allowed for this to be disabled. The following is the exact comments and code from the Robotc template:

```
  // Set bStopTasksBetweenModes to false if you want to keep user created tasks
  // running between Autonomous and Driver controlled modes. You will need to
  // manage all user created tasks if set to false.
  bStopTasksBetweenModes = true;
```

It would be great if this feature could be added as we would like to manage our own tasks on the V5 and not have them be killed by the Competition Switch...

HotelCalifornia, 2018-11-18:
Are you saying that your tasks _do_ end when you don't want them to?

PROS only manages the competition task (autonomous/opcontrol). We don't touch anything that users define, nor would it be really feasible for us to do so.

onthegrid007, 2018-11-19:
> Are you saying that your tasks _do_ end when you don't want them to?
> 
> PROS only manages the competition task (autonomous/opcontrol). We don't touch anything that users define, nor would it be really feasible for us to do so.

Sorry, I look like an idiot. I checked this a week ago when I was programming for a competition, and any user defined task seemed to be killed by any change using the Competition Switch. I was too busy to retest the issue or make a forum post until now, sorry for wasting your time and thanks for the quick response.

closed 2018-11-19 by onthegrid007