purduesigbots/pros-cli #21: Fails to build, intercept-c++: command not found error

dongwon900, 2018-11-20:
#### Expected Behavior:
I wanted to build an empty project to see if it works
#### Actual Behavior:
fails to build
#### Steps to reproduce:
prosv5 make command in the directory of the project
#### System information:
Operating System:  (Windows, OS X, Ubuntu)
OSX
PROS Version: (run `pros --version`)
3.1.2
#### Screenshots/Output Dumps/Stack Traces
-n Compiling src/autonomous.cpp 
-e \x1b[31;01m[ERRORS]\x1b[0m
/bin/sh: intercept-c++: command not found
make: *** [bin/autonomous.cpp.o] Error 1
Error: Failed to build

HotelCalifornia, 2018-11-21:
https://github.com/purduesigbots/pros-cli3/issues/15#issuecomment-439662168

HotelCalifornia, 2018-11-26:
This should be fixed with the release of 3.1.3. feel free to reopen if it is not.

closed 2018-11-26 by HotelCalifornia