purduesigbots/pros-cli #22: Docstrings out of date

Bottersnike, 2018-11-23:
The initial docstring I found that was out of date was https://github.com/purduesigbots/pros-cli3/blob/master/pros/serial/devices/vex/v5_device.py#L406, however I suspect there might be a few others dotted around the place that have likewise dropped out of date as code has changed over time (hence the vague issue title).

HotelCalifornia, 2018-12-04:
> `prosv5 c --help` directs users to this link which is broken:
`<https://pros.cs.purdue.edu/v5/cli/conductor>`
<sub>[View in Slack](https://pros-development.slack.com/archives/GAFDF0PEU/p1543899324000500)</sub>

edjubuh, 2019-01-31:
I'm not sure what the reported issue is, so closing for now until that's clear.

closed 2019-01-31 by edjubuh