purduesigbots/pros-cli #23: If field control is used, the robot has a probability of operating delay.

3038922, 2018-11-24:
Expected Behavior:
Complete a match normally.
Actual Behavior:
1 I confirm that my field control program is up to date.
If field control is used, the robot has a probability of operating delay.
The same problem occurs when the field control is not inserted, but the probability decreases.
Maybe I use NEW's relational system and occasionally it's not fully initialized. I'll add pros:: delay (100) to my test tomorrow.

Steps to reproduce:
System information:
Operating System: (Windows, OS X, Ubuntu)
windows 10 1807
PROS Version: (run pros --version)
pros, version 3.1.2

Additional Information
kernel 3.3.4
oklib 3.3.7

Screenshots/Output Dumps/Stack Traces
[image]

HotelCalifornia, 2018-11-24:
so you're saying that trying to use sensor values in a C++ constructor gives wrong values for the first ~100ms?

Where are you creating an instance of `Intake`? If it's in a global scope, then this is expected, as global C++ constructors are called before most things have finished initializing.

also: using `new` shouldn't have any appreciable effect on your program's runtime performance.

3038922, 2018-11-25:
Sorry,My student's judgment should be VEX OS problem. A robot never goes wrong. More than four V5 are the same opportunity to have this problem.

HotelCalifornia, 2018-11-26:
in that case, I'm going to close this issue as it doesn't seem to be PROS-specific.

closed 2018-11-26 by HotelCalifornia