purduesigbots/pros-cli #31: How to turn off GCC, a note type prompt?

3038922, 2018-12-19:
[image]
[image]

HotelCalifornia, 2018-12-19:
This is not an issue with PROS, but rather with C++ and GCC.

closed 2018-12-19 by HotelCalifornia

HotelCalifornia, 2018-12-19:
You can solve this [by adding `-Wno-psabi`](https://stackoverflow.com/a/48149400) to the `WARNFLAGS` variable in your Makefile

3038922, 2018-12-19:
> You can solve this [by adding `-Wno-psabi`](https://stackoverflow.com/a/48149400) to the `WARNFLAGS` variable in your Makefile
Thank you for testing his effectiveness.
I hope you will add this option by default in future versions.