purduesigbots/pros-cli #36: Packet parsing issue

Bottersnike, 2019-01-17:
#### Expected Behavior:
Received serial packets always parse

#### Actual Behavior:
Very rarely:
```py
ERROR - pros.serial.terminal.terminal:reader - unhashable type: 'bytearray'
  File "C:\Program Files\Python37\lib\site-packages\pros\serial\devices\vex\v5_user_device.py", line 37, in read
    while msg[0] is None or (msg[0] not in self.topics and not self._accept_all):
TypeError: unhashable type: 'bytearray'
```

#### Steps to reproduce:
Send an awful lot of packets until one eventually fails.

#### System information:
Operating System:  Windows 10 Enterprise

PROS Version: 3.1.3

#### Additional Information
I've solved the problem, however I haven't properly done it to a local branch to PR.

`v5_user_device.py` just requires ln49 changed to ```py
msg = bytes(msg[:4]), msg[4:]
```

The cast to `bytes()` allows it to used in the `not in` operator, whereas bytearrays cannot be used in this capacity.

edjubuh, 2019-01-31:
I haven't seen this issue pop up in any Sentry user reports and it seems you've just crashed it by fudging some messages. You're welcome to send in a PR to fix the issue if you'd like, but the team here won't open up the PR unless we run into the issue in the future.

edjubuh, 2019-02-01:
JK a bunch of people have run into this issue

PR #60 (2019-02-16): 🐛 Fix COBS packet parsing
#### Summary:
Explicitly convert bytearray to byte

#### Motivation:
58 reported events on Sentry

##### References (optional):
Closes #36

#### Test Plan:
I'm not sure how to test. I've never been able to repro the issue.
commit 9ebe45b4 (2019-02-16): Convert bytearray -> bytes (#60)

#### Summary:
Explicitly convert bytearray to byte

#### Motivation:
58 reported events on Sentry

##### References (optional):
Closes #36

#### Test Plan:
I'm not sure how to test. I've never been able to repro the issue.
closed 2019-02-16 by edjubuh