purduesigbots/pros-cli #37: cannot create project after installation

Windows10010, 2019-01-19:
#### Expected Behavior:
create a project
#### Actual Behavior:
errors creating the project with a return of `A project already exists in this location! Delete it first`
#### Steps to reproduce:
open the editor and click create a project
#### System information:
Operating System:  (Windows, OS X, Ubuntu)
OS X Mojave
PROS Version: (run `pros --version`)
3.1.3
#### Additional Information
The creation of a new project succeeded only once and took a long time to complete.
#### Screenshots/Output Dumps/Stack Traces
[image]

edjubuh, 2019-01-31:
After creating a project that fails, can you open the console and share any messages that were generated from you creating a new project? 

You can open the console by opening the command palette (Cmd+Shift+P I think on OS X) and searching for "Console: Toggle".

Windows10010, 2019-02-01:
The problem is solved. The cause was that I once carelessly created a project in the root directory in which almost all my files are stored(~/) and therefore new projects cannot be created in the subdirectories whose parent directory already contains a `project.pros`.

The directory diagram looks like this:

.
└── root
    ├── TheProjToBeCreatedIn
           └── WhereCreationFailed
    ├── makefilesAnd.etc
    ├── mydir1
    ├── mydir2
    ├── mydir3
    └── project.pros

When trying to create a project this way, pros CLI assumes that you are trying to create a project within one of the resources folders in a project already created. Such a fool I was.......`:(`

Apparently, this is a careless mistake of mine instead of a bug. But I suggest that It's good to add a confirmation message to the notification panel shown above that asks: `Are you trying to create a project in a project?`, so that users will get aware of the situation.

closed 2019-02-01 by Windows10010