purduesigbots/pros-cli #41: Request to implement a feature to add custom images for program on robot's brain

Bryanmarc, 2019-01-31:
**Desired Feature:**
Custom image for program on robot's brain

Bottersnike, 2019-01-31:
I may be wrong, but I believe these icons are hardcoded into VEXos and thus would require changes in the device firmware to allow for non-standard icons?

edjubuh, 2019-01-31:
As @Bottersnike mentioned, program icons are part of the assets package included in vexOS - they're not currently user-customizable. There are some user program icons included, but I'm not sure what they are. If someone is willing to experiment to figure out what icons exist and the relevant filenames, I'll add a feature to let you change it.

Relevant line:
https://github.com/purduesigbots/pros-cli/blob/develop/pros/serial/devices/vex/v5_device.py#L101

HotelCalifornia, 2019-01-31:
I think the original ask was actually less about arbitrary icons and more about letting users select from the ones that are built-in (à la VCS)

Bottersnike, 2019-01-31:
If I remember correctly, the icons go from USER901x to USER905x, but there might be others I didn't find when I went hunting a while back.

Nenik, 2019-10-23:
I've got pizza slice for USER003x at some point :-)

Octopus118, 2020-12-10:
A list of icons available: https://www.vexforum.com/t/a-guide-to-changing-program-icons/78293/7

kunwarsahni01, 2022-02-02:
Now added in #194

closed 2022-02-02 by kunwarsahni01