purduesigbots/pros-cli #52: Upload hanging or Device or Resource Busy on Linux

kusti8, 2019-02-10:
#### Expected Behavior:
For pros to upload the project to the v5 brain.

#### Actual Behavior:
Sometimes, I get a Device or Resource Busy message or the upload just hangs at some random percentage and won't continue. Sometimes running `udevadm trigger` and unplugging and replugging works, but other times it just hangs.

I am part of the dialout group and this only happens occasionally. If I spin up a Windows VM right after it hangs, and connect that to the USB device, then the upload works without issue. If I reboot and try again, the problem persists.

#### Steps to reproduce:
1. Plug in brain and try to upload with `pros v5 u`.

#### System information:
Operating System: Ubuntu

PROS Version: 3.1.3

#### Additional Information

#### Screenshots/Output Dumps/Stack Traces
```
make: Nothing to be done for 'quick'.
Uploading bin/output.bin to v5 device on /dev/ttyACM0 as States to slot 1
ERROR - pros.cli.upload:upload - [Errno 16] could not open port /dev/ttyACM0: [Errno 16] Device or resource busy: '/dev/ttyACM0'
  File "/home/kusti8/.local/lib/python3.6/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 16] could not open port /dev/ttyACM0: [Errno 16] Device or resource busy: '/dev/ttyACM0'
```
After running `udevadm trigger` and replugging the wire, it hangs
```
make: Nothing to be done for 'quick'.
Uploading bin/output.bin to v5 device on /dev/ttyACM0 as States to slot 1
Uploading slot_1.ini [####################################] 100%
Uploading slot_1.bin (bin/output.bin) [##########--------------------------]  27%
```

HotelCalifornia, 2019-02-17:
closing as a duplicate of #45

kusti8, 2019-02-17:
I saw the other issue, however mine seems to be slightly different. Mine hangs midway through downloading, and I do have permissions, and there are no other processes that are running as far as I can tell (I ran lsof and tried to kill any other python scripts).

edjubuh, 2019-02-17:
Can you run `prosv5 u --debug > debug.output.txt` and share it via gist or pastebin or upload file?

kusti8, 2019-02-17:
https://pastebin.com/Kack4rrN

https://pastebin.com/2NgwJYSt

edjubuh, 2019-02-17:
🤔 you're on the develop branch? Just making sure we're on the same page

kusti8, 2019-02-17:
I switched to the wireless upload branch to test that out, but the issue was the same on the master branch and the develop branch.

edjubuh, 2019-02-17:
I might be able to patch something for this log, but I'm still not sure why it failed half-way through as you originally posted since the log you attached shows it failed at an earlier stage.

edjubuh, 2019-02-17:
I've created a branch: [hotfix/kusti8-debugging](https://github.com/purduesigbots/pros-cli/tree/hotfix/kusti8-debugging) that has a potential patch for the log you sent.

kusti8, 2019-02-17:
The point at which it fails is different every time, usually stopping earlier every time. I'll try out the fix, thanks.

edjubuh, 2019-02-17:
I noticed you ctrl+c'd it out. You did say it was hanging, which is odd because we set timeout to something like 1 second with 3 retries of the same command, so it should never hang for more 3-ish seconds. I assume it hangs for 10+ seconds?

kusti8, 2019-02-17:
Yeah it hangs for 10+ seconds. I had to Ctrl+c it.

kusti8, 2019-02-18:
Hi, the patch prevents it from hanging. It still says device or resource busy, so I have to run `udevadm trigger` 1 or 2 times, but it works consistently after that.

edjubuh, 2019-02-18:
Glad to hear it works. I'll close this issue for now unless you can determine a consistent workflow that causes the "device or resource busy" error. We'd have to report that to upstream pyserial.

closed 2019-02-18 by edjubuh