purduesigbots/pros-cli #79: pros won't work

pieceofpie, 2019-02-22:
#### Expected Behavior:

the command would not run after installed the new version of pros 3.1.4.0.

#### Actual Behavior:

cannot find the files specified. See the attached picture for detail.

#### Steps to reproduce:

#### System information:
Operating System:  
OS Version:	Windows 10
OS_Architecture:	64-bit
Service Pack:	No service pack
CPU:	Quad 3.70 GHz Intel Xeon(R)E3-1240 v6 (Hyper-Threaded)
Memory:	32700 MB
Manufacturer:	Dell Inc.
Model:	Precision Tower 3620
Service Tag:	6F101S2

PROS Version: 
pros 3.1.4.0

#### Additional Information

#### Screenshots/Output Dumps/Stack Traces
[image]

edjubuh, 2019-02-22:
Do you have the PROS_TOOLCHAIN environment variable set? The installers should do this, but sometimes they don't on some machines.

pieceofpie, 2019-02-22:
C:\Users\xyz>echo %PROS_TOOLCHAIN%
C:\Users\xyz\AppData\Local\Programs\PROS\toolchain\

find subprocess.py in C:\Users\xyz\AppData\Local\Programs\Python\Python37-32\Lib\

pieceofpie, 2019-02-25:
solve the issue after uninstall and manually remove the residual files. reinstall PROS and everything seems fine

closed 2019-02-25 by HotelCalifornia