purduesigbots/pros-cli #80: Linux build not finding cerrno

Minichota, 2019-02-22:
#### Expected Behavior:
I would expect to be able to compile my programs

#### Actual Behavior:
I get an error: ./include/api.h:23:18: fatal error: cerrno: No such file or directory
 #include <cerrno>

#### System information:
Linux Mint

PROS Version: 
Version 3.1.4
#### Additional Information
The error is produced when I run make.

#### Screenshots/Output Dumps/Stack Traces

Octogonapus, 2019-02-23:
Can you post the output of `arm-none-eabi-gcc --version`?

Minichota, 2019-02-23:
> Can you post the output of `arm-none-eabi-gcc --version`?

command not found, if I just do gcc --version I get: 7.3.0

Octogonapus, 2019-02-23:
Well, that's the problem. You don't have the toolchain installed. Please install the latest version using your favorite package manager or from the ARM developer site [here](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads).

Minichota, 2019-02-23:
> Well, that's the problem. You don't have the toolchain installed. Please install the latest version using your favorite package manager or from the ARM developer site [here](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads).

I installed it to no success.  I still get the same error.

Octogonapus, 2019-02-23:
That version is too old. Please grab at least 7.3.

HotelCalifornia, 2019-02-23:
Please see step 1 of the [getting started guide for Linux](https://pros.cs.purdue.edu/v5/getting-started/linux.html).

Octogonapus, 2019-02-23:
> Please see step 1 of the [getting started guide for Linux](https://pros.cs.purdue.edu/v5/getting-started/linux.html).

Fair, but I linked them the ARM site directly because they run Mint.

Minichota, 2019-02-23:
I installed a newer version and verified it. I have it all downloaded but the files in the bin directory are still the old version. What is the best method to move such things from one place to another or I guess, upgrade?

Minichota, 2019-02-23:
Should I just copy the files over from the bin directory of the install?

edjubuh, 2019-02-23:
> Should I just copy the files over from the bin directory of the install?

No. You should extract gcc-arm-none-eabi-8-2018-q4-major-linux.tar.bz2 to somewhere like `/opt/gcc-arm-none-eabi` (anywhere is fine, so long as you don't accidentally delete) and then add `/opt/gcc-arm-none-eabi/bin` to your path.

HotelCalifornia, 2019-02-23:
Mint is a debian-based distribution. Just install the apt PPA linked in the getting started guide and save yourself the hassle of a manual install.

Minichota, 2019-02-23:
upon trying to install using the ppa I get this error:
Errors were encountered while processing:
 /var/cache/apt/archives/gcc-arm-embedded_7-2018q2-1~bionic1_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

HotelCalifornia, 2019-02-23:
>If it reports error message of conflict to gcc-arm-none-eabi, which is likely if upgrading from 4.x to 5+, please uninstall it first with:
>       "sudo apt-get remove gcc-arm-none-eabi"

Taken from [here](https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa)

Minichota, 2019-02-24:
> > If it reports error message of conflict to gcc-arm-none-eabi, which is likely if upgrading from 4.x to 5+, please uninstall it first with:
> > "sudo apt-get remove gcc-arm-none-eabi"
> 
> Taken from [here](https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa)

I have it uninstalled already and get the same error when trying to install the newer version.

HotelCalifornia, 2019-02-24:
Yeah, you probably also need to remove the version that you installed manually

Minichota, 2019-02-24:
After uninstalling the version I had installed and the manual install version I still get virtually the same error:
dpkg: error processing archive /var/cache/apt/archives/gcc-arm-embedded_7-2018q2-1~bionic1_amd64.deb (--unpack):
 trying to overwrite '/usr/bin/arm-none-eabi-addr2line', which is also in package binutils-arm-none-eabi 2.27-9ubuntu1+9
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/gcc-arm-embedded_7-2018q2-1~bionic1_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

Octogonapus, 2019-02-24:
Even if you think you totally uninstalled it, we can see that it has not been completely uninstalled. If you believe all the files that got installed have been removed, then just remove the `arm-none-eabi-*` files in `/usr/bin/`.

HotelCalifornia, 2019-02-25:
I'm going to close this issue for now as it isn't specifically a problem with PROS.

closed 2019-02-25 by HotelCalifornia