purduesigbots/pros-cli #88: latest Pros 3.1.4.0 32bit exe installs old cortex pros

Samuel12321, 2019-03-08:
#### System information:
Operating System:  
windows 10 1803 32bit
sorry will add more info soon

HotelCalifornia, 2019-03-13:
any more info?

Samuel12321, 2019-03-13:
Seems to install the pros 0.75 package and bootstrapper

Samuel12321, 2019-03-13:
This causes an error stating it cant find the pros cli on path,
possibly due to having the pros cli 3 and the pros 0.75 packages

Samuel12321, 2019-03-15:
Will try to test again on another 32bit laptop running windows 7. Is there anything you want me to do to make the test more usefull?

kunwarsahni01, 2021-05-14:
We have now removed all 32-bit installers so this is no longer applicable.

closed 2021-05-14 by kunwarsahni01

HotelCalifornia, 2021-05-14:
you may still use the CLI directly on 32-bit machines by extracting the standalone CLI release (e.g. pros_cli-3.2.1-win-32bit.zip), however you will not get the full PROS Editor experience (and we will likely not be able to help if there are any problems)