purduesigbots/pros-cli #93: flash() got an unexpected keyword argument 'file'

hghowe, 2019-04-18:
#### Expected Behavior:
upload a built program to robot (Cortex)

#### Actual Behavior:
Does not upload. I get a message:
TypeError: flash() got an unexpected keyword argument 'file'

#### Steps to reproduce:
Happens every time I try to upload. Program is the base level program, and it built successfully.

#### System information:
Operating System:  
OSX 10.13.6

PROS Version: 
2.6.1

#### Additional Information

Newly installed Atom & PROS. Robot connected via computer to programming kit to joystick to USB to VEX cortex.

#### Screenshots/Output Dumps/Stack Traces
US-Robotics:PROS_Test1 kadmin$ pros flash -vd -f "/Users/kadmin/Documents/PROS_Test1"
Traceback (most recent call last):
  File "/usr/local/bin/pros", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/site-packages/proscli/main.py", line 9, in main
    cli.main(prog_name='pros')
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 1137, in invoke

    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/click/decorators.py", line 64, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
TypeError: flash() got an unexpected keyword argument 'file'

hghowe, 2019-04-23:
Just loaded a clean install of PROS on another mac. Same error.

edjubuh, 2019-04-23:
It's possible that specifying a binary file broke at some point. We're no longer developing PROS CLI 2 (in favor of PROS CLI 3, which supports both Cortex and V5 projects).

As a workaround, don't specify the file. If you're in a PROS project directory, the CLI will figure out what to upload. 

If you're insistent that you need to specify `-f`, you should modify `proscli/flasher.py` line 35 (I'm not sure where this file would be on a Mac - @HotelCalifornia  or @jwalk511 might be able to help with that)  and change `binary` to `file` and see if that works for you.

hghowe, 2019-04-24:
Also tried it with another Cortex. No change.

HotelCalifornia, 2019-04-26:
Did you try any of @edjubuh's suggestions?

hghowe, 2019-05-29:
Strangely, I hadn't seen his post when I posted on the 24th. We were trying to use the Cortex for a specific project - we just gave up and pulled the V5 from our competition bot. Then the end of the school year came along, and I got tied up in that. 

I had just been using the GUI - I never specified a file, to my knowledge. I had just pressed the upload button, and that is what it scripted. 

It sounds like the best thing to do is to switch over to PROS CLI 3.

I may just give up on this thread - I was using some of our old Cortexes (Cortices?) for a recruitment project - we're mostly developing for V5 these days.

closed 2019-11-16 by HotelCalifornia