purduesigbots/pros-cli #94: Feature request for terminal over wireless

lerker100, 2019-05-05:
#### Expected Behavior:
When plugged into the V5 controller to be able to run prosv5 terminal and see the terminal output of the brain.

#### Actual Behavior:
Gives an error that you need to plug in to the brain.

#### Steps to reproduce:
1. Plug in to a V5 controller paired to a V5 brain.
2. Run a program that has terminal output via printf.
3. open a terminal on your computer and run prosv5 terminal.

#### System information:
Platform:V5
PROS Kernel Version: 3.1.6

edjubuh, 2019-05-05:
What's the output of `prosv5 lsusb --debug`?

lerker100, 2019-05-05:
I don't have a brain to test with until tomorrow but it was connected because a could upload over the connection

edjubuh, 2019-05-05:
If you're connected via joystick, we don't support user terminal over the joystick yet.

lerker100, 2019-05-05:
That's why I labeled it as a feature request. It would be a nice to have for debugging.

edjubuh, 2019-05-05:
Oh pff, totally didn't read the title or the first phrase of the issue.

lerker100, 2019-05-05:
Is there a better place for feature requests?

HotelCalifornia, 2019-05-05:
No, here is fine

lerker100, 2019-05-05:
Any luck with the vision sensor? I was told back in February that you all hadn't gotten it to work yet.

HotelCalifornia, 2019-05-06:
I'm not sure what you mean, but let's stay on topic here

HotelCalifornia, 2020-09-15:
This feature hasn't landed yet, not sure why we're closing it

lerker100, 2020-09-15:
The PR said that it had been merged. I figured that GitHub just didn't close this issue for some reason.

PR #114 (2019-12-18): Fix wireless terminal (RX only)
#### Summary:
Fixed user_fifo_read by properly formatting the data required by the ext 0x27 command so that it would not fail with a General NACK, then adding a `_serial_cache` so that the terminal would only receive full COBS-decodable messages and not output "Could not decode bytes".

#### Motivation:
This change brings wireless terminal to PROS, a feature that's been present in VEXCode and RMS. This will allow for remote debugging of PROS programs.

##### References (optional):
This closes #94.

#### Test Plan:
- [x] Be able to read from terminal
- [ ] Be abl [... cut]
closed 2021-05-14 by kunwarsahni01