purduesigbots/pros-cli #95: The output file become bigger when you upgrade your kernel to 3.1.6

Jerrylum, 2019-05-15:
The BBS file is to big and slow down the upload time.

#### Steps to reproduce:
1. make a folder call prosv5-sandbox3
2. open a new console on there
3. use command `prosv5 c n .`
4. Because of auto compile, you can see the result of section size, it is about 48MB
[image]

5. copy `common.mk`, `compile_commands.json` and `Makefile` to a folder call "backup"
6. use command `prosv5 c upgrade kernel@3.1.4`
7. paste the file from `backup` folder to `prosv5-sandbox3`
8. use command `prosv5 make all` and you can see it is only 3MB
[image]

#### System information:
Operating System:  Windows

PROS Version: 3.1.6

#### Additional Information
It may not a problem of pros-cli because I try to reproduce the issue on cli@3.1.2 and it has the same result.

HotelCalifornia, 2019-05-15:
https://en.wikipedia.org/wiki/.bss

Jerrylum, 2019-05-15:
I know BSS doesn't take up any actual space in the object file now and it won't slow down the upload time, but I still want to know why it is very big. What's the difference between two version?

edjubuh, 2019-11-16:
> I know BSS doesn't take up any actual space in the object file now and it won't slow down the upload time, but I still want to know why it is very big. What's the difference between two version?

We increased heap size from 1MB to 48MB
https://github.com/purduesigbots/pros/commit/728dcb1a4f8bb693bf3b71d6d0369033bad848ab#diff-3e67e7437ae7de6cc9138170266edcdaR1
https://github.com/purduesigbots/pros/commit/728dcb1a4f8bb693bf3b71d6d0369033bad848ab#diff-4bf621f08c857fdd0610fb8794f275e2L17

closed 2019-11-16 by edjubuh