purduesigbots/pros-cli #96: Unable to upload without PROS project

lights0123, 2019-05-18:
#### Expected Behavior:
As per the help in `prosv5 upload --help`:

>   [PATH] may be a directory or file. If a directory, finds a PROS project root and uploads the binary for the correct target automatically. **If a file, then the file is uploaded**. Note that --target must be specified in this case.

Attempting to upload a file using `prosv5 upload --target v5 path/to/binary.bin` without a PROS project should work.
#### Actual Behavior:
The following error message is given:
```
❱ prosv5 upload --target v5 path/to/binary.bin
Usage: pros upload [OPTIONS] [PATH] [PORT]

Error: «current directory» is not inside a PROS project. Execute this command from within a PROS project or specify it with --project project/path
```

#### Steps to reproduce:
1. Go to a directory that does not have a `project.pros` file, and have a V5 system connected
2. Place a valid binary in an accessible path, e.g. with `objcopy -O binary -R .hot_init` or copying from a PROS project
3. Attempt to upload a file with `prosv5 upload --target v5 path/to/binary.bin`

#### System information:
Operating System:  Antergos Linux

PROS Version: `pros, version 3.1.4`

HotelCalifornia, 2019-05-19:
Yeah I'm pretty sure the problem is that V5 uploads also require the .ini file, and I'm not sure that gets generated automatically when specifying a file

HotelCalifornia, 2019-05-19:
Which is by design-- we can't really guess values for the content without project info. 

(I'm assuming here that you're trying to upload a binary that will be run as user code, which is what the `prosv5 u` command is for)

HotelCalifornia, 2019-09-19:
@lights0123 you were trying to upload a binary to be run as user code, right?

I think the best we can do here is to add logic that won't accept a file when the target is v5 and some to be determined set of the v5 options flags are left unspecified.

HotelCalifornia, 2019-11-16:
In the meantime, you can work around this issue by creating a dummy project.pros file in your project directory, then specifying the path to the binary.

PR #116 (2021-02-18): Fix uploading without a project
#### Summary:
Bubbled up a None project if we can't find it and say we don't need it. I also added some additional verification I noticed was needed to make sure a target was specified. I haven't had a chance to test this, so I might also need to do something with the .ini files in the upload layer.

#### Motivation:
A couple of people have mentioned wanting this, and it's convenient occasionally.

##### References (optional):
Fixes #96 

#### Test Plan:

- [x] Uploading to a v5 still works with a project
- [x] Uploading to a v5 works without a project

#### Ope [... cut]
closed 2021-02-18 by HotelCalifornia