purduesigbots/pros-cli #97: PROS CLI not found by PROS Editor

baylessj, 2019-05-29:
#### Expected Behavior:
Atom will detect the presence of the CLI and not give the welcome screen with a "CLI Missing" error upon opening the program.

#### Actual Behavior:
CLI is installed and on PATH but the PROS welcome screen says that the CLI is not installed every time I open atom.

#### Steps to reproduce:
Ran through the installation steps on the docs site, issue has been persistent since.

#### System information:
Operating System:  Pop!_OS 19.04

PROS Version: 3.1.4

HotelCalifornia, 2019-05-29:
It seems like you have the old version of the atom plugin, as we have not yet added that screen to the new one. Just to confirm, you followed this step when installing?

>3. Run apm install pros-bootstrapper@0.0.12

baylessj, 2019-05-29:
I ran that command but somehow had the old pros package installed on my machine as well. Not sure how I managed to get the old pros package installed, but removing it solved the issue.

closed 2019-05-29 by baylessj