purduesigbots/pros-cli #98: Add progress reporting for new project creation

Octogonapus, 2019-06-11:
#### Expected Behavior:
New project creation should report its progress (even if that is just "in progress" and "complete").

#### Actual Behavior:
New project creation is silent.

#### Steps to reproduce:
1. Create a new project in the PROS Editor.
2. The editor does nothing until the new project is fully created.

#### System information:
Operating System:  macOS

PROS Version: ¯\_(ツ)_/¯

HotelCalifornia, 2019-06-11:
Rather than progress, a simply busy spinner should suffice I think. Shouldn't be too hard to add as an interactive UI component (assuming there isn't already one; I can't remember off the top of my head).

For what it's worth, the template acquisition (if necessary) and application to a new project should already report progress in notifications.

Octogonapus, 2019-06-12:
> For what it's worth, the template acquisition (if necessary) and application to a new project should already report progress in notifications.

Template acquisition works as expected.

kunwarsahni01, 2021-05-14:
This is a plugin bug on macOS, new project creation does show reporting on windows.