purduesigbots/pros-cli #99: macOS Terminal error decoding

Octogonapus, 2019-06-12:
Not sure if this is helpful. Brad ran into this running `prosv5 mut`:

WARNING - pros.serial.devices.vex.v5_user_device:read - Could not decode bytes: 202020202020201b5b313b33336d221b5b306d20202020202020202020202020202020436f70797269676874202863292050757264756520556e69766572736974792041434d20534947426f74730a

HotelCalifornia, 2019-06-12:
Uh. Can you check what locale is set?

Octogonapus, 2019-06-12:
You mean in the PROS Editor? We were using vscode. I think we can follow these instructions https://code.visualstudio.com/docs/getstarted/locales

HotelCalifornia, 2019-06-12:
Ah, I assumed you were just using the default terminal app.

Octogonapus, 2019-06-13:
The locale is `en`

Octogonapus, 2019-06-15:
@HotelCalifornia Anything else I should look at?

HotelCalifornia, 2019-06-16:
Sorry, I really have no clue what's going on here.

Octogonapus, 2019-06-16:
I think we can at least do better in this case. Those above bytes parse into the string `       [1;33m"[0m                Copyright (c) Purdue University ACM SIGBots`. I would like to see that get printed instead of having the terminal crash.

jophy-ye, 2019-07-22:
Well, I am using Mac High Sierra. I have the same issue too. With the exact same message:

WARNING - pros.serial.devices.vex.v5_user_device:read - Could not decode bytes: 202020202020201b5b313b33336d221b5b306d20202020202020202020202020202020436f70797269676874202863292050757264756520556e69766572736974792041434d20534947426f74730a

Octogonapus, 2019-07-22:
@HotelCalifornia How do you feel about implementing best-effort parsing when the normal parsing fails?

HotelCalifornia, 2019-07-22:
Do you mind elaborating on what you had in mind?

Octogonapus, 2019-07-22:
> Do you mind elaborating on what you had in mind?

Like what I said above, that string of bytes can be decoded into `�[1;33m"�[0m Copyright (c) Purdue University ACM SIGBots`. I would much rather see corruption in the terminal output than have the terminal crash. The exception comes from here, when `cobs.decode(msg)` fails, right? https://github.com/purduesigbots/pros-cli/blob/9ebe45b4ad971367641e386b546a9f23ec7ef492/pros/serial/devices/vex/v5_user_device.py#L46

We should at least be able to parse the raw bytes into a string (that's how I got to `�[1;33m"�[0m Copyright (c) Purdue University ACM SIGBots`). Someone with more knowledge of cobs might be able to implement a smarter parser to recover more information.

HotelCalifornia, 2019-07-22:
Any chance you can try

1. in a regular (Terminal.app etc.) terminal? 
2. with the debug flag enabled, so we can get a stack trace?

Octogonapus, 2019-07-22:
Not until at least two weeks from now.

HotelCalifornia, 2019-07-22:
Sorry, didn't catch the line you linked above. It looks like the `cobs.DecodeError` is getting caught by this fallthrough exception handler. Will be straightforward to add an explicit case for this that doesn't end the process like we do for unicode decode failures.
https://github.com/purduesigbots/pros-cli/blob/9ebe45b4ad971367641e386b546a9f23ec7ef492/pros/serial/terminal/terminal.py#L243-248

jophy-ye, 2019-07-23:
@HotelCalifornia When I was using "pros terminal", I used a regular terminal (Terminal.app). I might run with the debug flag later

Christopher-Tan, 2022-10-23:
Cannot reproduce the error

closed 2022-10-23 by Christopher-Tan