purduesigbots/pros-cli #101: Failed to create a new project

berberman, 2019-07-13:
#### Expected Behavior:
A new v5 project can be created and compiled successfully.

#### Actual Behavior:
Failed to compile the project.  
Errors arose in linking project with libpros,okapilib which are caused by `__sync_synchronize`.

#### Steps to reproduce:
```bash
> prosv5 conductor new-project ~/Desktop/hello v5
```

#### System information:
Operating System:  Manjaro 18.0.4 Illyria

PROS Version: 3.1.4

#### Additional Information
I only installed PROS CLI and GNU Arm Embedded Toolchain (I built it manually).

#### Screenshots/Output Dumps/Stack Traces
```
Applying  kernel@3.1.6
Finished  applying  kernel@3.1.6  to  hello
Applying  okapilib@3.3.13
Finished  applying  okapilib@3.3.13  to  hello
New  PROS  Project  was  created:
PROS  Project  for  v5  at:  /home/berberman/Desktop/hello  (hello)
Name            Version        Origin
--------    ---------    -------------
kernel        3.1.6            pros-mainline
okapilib    3.3.13          pros-mainline
Building  project...
Compiling  src/autonomous.cpp  [OK]
Compiling  src/initialize.cpp  [OK]
Compiling  src/opcontrol.cpp  [OK]
Adding  timestamp  [OK]
Linking  project  with  libpros,okapilib  [ERRORS]
/usr/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld:  /usr/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/lib/libstdc++.a(locale.o):  in  function  `std::locale::_Impl::_M_install_cache(std::locale::facet  const*,  unsigned  int)':
locale.cc:(.text._ZNSt6locale5_Impl16_M_install_cacheEPKNS_5facetEj+0x18):  undefined  reference  to  `__sync_synchronize'
/usr/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld:  /usr/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/lib/libstdc++.a(cxx11-ios_failure.o):  in  function  `(anonymous  namespace)::__io_category_instance()':
cxx11-ios_failure.cc:(.text._ZN12_GLOBAL__N_122__io_category_instanceEv+0xc):  undefined  reference  to  `__sync_synchronize'
/usr/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld:  /usr/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/lib/libstdc++.a(locale_init.o):  in  function  `(anonymous  namespace)::get_locale_mutex()':
locale_init.cc:(.text._ZN12_GLOBAL__N_116get_locale_mutexEv+0xc):  undefined  reference  to  `__sync_synchronize'
collect2:  error:  ld  returned  1  exit  status
```

Octogonapus, 2019-07-13:
Please install a version equal to 8.2.x or lower. Something is wrong with 8.3.x.

HotelCalifornia, 2019-07-13:
[Here is the official link to the toolchain](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads)

berberman, 2019-07-20:
> Please install a version equal to 8.2.x or lower. Something is wrong with 8.3.x.

Thanks for your help! I installed 8.2.1 and now it can be compiled correctly :wink:

closed 2019-07-20 by berberman