purduesigbots/pros-cli #102: Linking fails with Arm Toolchain 8-2019-q3-update Linux 64-bit

aderic6, 2019-07-28:
#### Expected Behavior:
Ran prosv5 make all and expected to compile into hot and cold image.

#### Actual Behavior:
Compiling of source files works until attempting to link with libpros and libokapi. The same set of files compile with the previous 2018 December Arm Toolchain.

#### Steps to reproduce:
1. Download the new Linux 64-bit toolchain from [Arm Developers](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads)
2. Extract and move the folder to /opt/arm-gcc-none-eabi/ replacing old toolchain
3. Export to PATH using ~/.profile and sourcing the file.
4. Make a new PROS project with conductor.
5. Run prosv5 make in project directory (both with and without hot and cold linking yields the same error).

#### System information:
Manjaro 18.0.4 (4.19.60-1-MANJARO)

PROS Version: pros, version 3.1.4

#### Additional Information
Ran on a clean install with no other conflicting packages and whereis points to the extracted toolchain. The linking failure occurs with the default PROS project source files too.

#### Screenshots/Output Dumps/Stack Traces
[image]

HotelCalifornia, 2019-07-28:
Thanks for the report, I was worried something like this would happen when I saw Arm updated the official toolchain. Will need to investigate to see what's wrong.

Octogonapus, 2019-07-28:
Use version 8.2.1 or lower.

HotelCalifornia, 2019-07-28:
>Use version 8.2.1 or lower.

That was mentioned as a workaround in the OP. We still need to figure out what's wrong.

Octogonapus, 2019-07-28:
> That was mentioned as a workaround in the OP

My bad, I am so used to replying to these `__sync_synchronize` issues...