purduesigbots/pros-cli #104: Problems with simultaneous installation of py2 py3.74 on win10

3038922, 2019-09-14:
run "prosv5 build-compile-commands"

```
PS D:\towerTakeOver> prosv5 build-compile-commands
We detected something went wrong! Do you want to send a report? [y/N]: N
Not sending bug report.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "c:\users\aresp\appdata\local\programs\python\python37-32\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "c:\users\aresp\appdata\local\programs\python\python37-32\lib\site-packages\sentry_sdk\integrations\threading.py", line 47, in run
    reraise(*_capture_exception())
  File "c:\users\aresp\appdata\local\programs\python\python37-32\lib\site-packages\sentry_sdk\_compat.py", line 52, in reraise
    raise value
  File "c:\users\aresp\appdata\local\programs\python\python37-32\lib\site-packages\sentry_sdk\integrations\threading.py", line 45, in run
    return old_run(*a, **kw)
  File "c:\users\aresp\appdata\local\programs\python\python37-32\lib\site-packages\pros\common\ui\__init__.py", line 180, in run
    for line in iter(self.pipeReader.readline, ''):
UnicodeDecodeError: 'gbk' codec can't decode byte 0x80 in position 10: illegal multibyte sequence

PS D:\towerTakeOver>
```

run pip3 list
```
PS D:\towerTakeOver> pip3 list
Package          Version
---------------- -----------
cachetools       3.1.0
certifi          2018.11.29
chardet          3.0.4
click            6.7
cobs             1.1.3
colorama         0.4.1
idna             2.8
jsonpickle       1.1
LEPL             5.1.3
observable       1.0.3
pip              19.2.3
pros-cli-v5      3.1.4
pypng            0.0.19
pyserial         3.4
pyzmq            18.0.0
requests         2.21.0
rfc6266-parser   0.0.5.post2
scan-build       2.0.13
semantic-version 2.6.0
sentry-sdk       0.7.4
setuptools       40.8.0
tabulate         0.8.3
typing           3.6.6
urllib3          1.24.1
```

HotelCalifornia, 2019-09-14:
Since it's only on your computer, and by your own admission the development environment is complicated, I'm not sure how this is a problem with the PROS CLI?

3038922, 2019-09-15:
It's OK to manually change the compile_commands file.

closed 2019-09-15 by 3038922