purduesigbots/pros-cli #107: Template upgrade notificatoins

HotelCalifornia, 2019-09-19:
The CLI should notify the user when an upgrade for a template in a project becomes available. This should be debounced so the notifications don't appear too often.

lerker100, 2020-04-12:
I think this function would do most of what is needed.
https://github.com/purduesigbots/pros-cli/blob/b4437e8d592a31bac0d6b922f1788ff924b3cb77/pros/conductor/project/__init__.py#L92-L93

There's also a list of current templates. 
https://github.com/purduesigbots/pros-cli/blob/b4437e8d592a31bac0d6b922f1788ff924b3cb77/pros/conductor/interactive/UpdateProjectModal.py#L70

I would like to try to take this on but not quite sure how to get these things put together.

HotelCalifornia, 2020-04-12:
So, I'm thinking the logic would be more or less the same as the way we use `Depot.last_remote_update` and `Depot.update_frequency`: 

https://github.com/purduesigbots/pros-cli/blob/80684a1d835a474f19bc417d4842f32b90a8b8ca/pros/conductor/depots/depot.py#L29-L37

the main difference being that we'd want this check to be specific to each project. The one part I haven't quite thought through yet is when to trigger a check for upgradable templates. Whenever the project is built maybe? @edjubuh maybe you want to chime in on that

lerker100, 2020-04-12:
When the project gets built is what I was thinking as well. Right after the comment.
https://github.com/purduesigbots/pros-cli/blob/b4437e8d592a31bac0d6b922f1788ff924b3cb77/pros/cli/build.py#L19-L27