purduesigbots/pros-cli #108: Add File Redirect Option to Terminal

py2025, 2019-09-22:
User should be able to select the file that they want terminal output to be directed to.

HotelCalifornia, 2019-09-22:
something like `prosv5 terminal --output filename`. I think we'd also want it to still display to the user what's going on (while copying output to the file in the background)

3038922, 2019-09-23:
It would be great if we could interact more conveniently with the host computer.

Personally, I'm going to start learning tensorflow in the near future.

edjubuh, 2019-09-23:
There's already standard shell utilities like `prosv5 terminal > output.txt` or `prosv5 terminal | tee output.txt`. Why add this to PROS?

HotelCalifornia, 2019-09-23:
Because it isn't hard to add and I feel that it improves user experience. People who already know how to use those standard shell utilities can of course continue to use them.

lerker100, 2020-04-21:
If your using bash you can just run this `prosv5 terminal > test.txt`

PR #220 (2022-08-22): ✨Terminal file output
#### Summary:
Added terminal option that sends output to a file by redirecting the standard output stream. Output is displayed in both the terminal and file. 

##### References (optional):
Closes #108 (Add File Redirect Option to Terminal)

#### Test Plan:
Using the command `pros terminal --output [filename]` should show output in the terminal and the selected file. Afterwards, it should return to the original state of the standard output stream (terminal only).
closed 2022-08-22 by WillXuCodes