purduesigbots/pros-cli #110: Upload doesn't honour project-defined slot

Nenik, 2019-10-23:
#### Expected Behavior:
"prosv5 upload" with no "--slot" would upload to the slot specified in project options.

#### Actual Behavior:
"prosv5 upload" with no "--slot" always uploads to the first slot

#### Steps to reproduce:
Create a new project.
Edit project.pros, add **"slot": 4** to the upload_options section.
Issue prosv5 mu

#### System information:
Operating System:  Arch Linux (the same was reproduced on Windows 7).

PROS Version: cli: 3.1.4, kernel: 3.20

#### Additional Information
The issue is caused by the python/click command line parsing subsystem and the fact that the hardcoded default is delivered the same way as the explicit command line option to the upload() function. In fact, the expected order of precedence of the slot selection would be:

1. Command Line
2. project.pros upload_options
3. Built-in/hard-coded default.

#### Screenshots/Output Dumps/Stack Traces
Possible solution (if ugly):

```
diff --git a/pros/cli/upload.py b/pros/cli/upload.py
index b592988..980a56f 100644
--- a/pros/cli/upload.py
+++ b/pros/cli/upload.py
@@ -19,7 +19,7 @@ def upload_cli():
 @click.option('-q', '--quirk', type=int, default=0)
 @click.option('--name', 'remote_name', type=str, default=None, required=False, help='Remote program name',
               cls=PROSOption, group='V5 Options')
-@click.option('--slot', default=1, show_default=True, type=click.IntRange(min=1, max=8), help='Program slot on the GUI',
+@click.option('--slot', type=click.IntRange(min=1, max=8), help='Program slot on the GUI',
               cls=PROSOption, group='V5 Options')
 @click.option('--program-version', default=None, type=str, help='Specify version metadata for program',
               cls=PROSOption, group='V5 Options', hidden=True)
@@ -57,9 +57,13 @@ def upload(path: Optional[str], project: Optional[c.Project], port: str, **kwarg
             kwargs['remote_name'] = project.name
 
         # apply upload_options as a template
+        if kwargs['slot'] == None:
+            del kwargs['slot']
         options = dict(**project.upload_options)
         options.update(kwargs)
         kwargs = options
+        if 'slot' not in kwargs:
+            kwargs['slot'] = 1
 
         kwargs['target'] = project.target  # enforce target because uploading to the wrong uC is VERY bad
         if 'program-version' in kwargs:
```

edjubuh, 2019-11-03:
I guess `options.update(kwargs)` was too much of a shortcut. This ought to do better (only slightly tested):

```diff
diff --git a/pros/cli/upload.py b/pros/cli/upload.py
index b592988..e1e3ec9 100644
--- a/pros/cli/upload.py
+++ b/pros/cli/upload.py
@@ -15,11 +15,11 @@ def upload_cli():
 @click.argument('path', type=click.Path(exists=True), default=None, required=False)
 @click.argument('port', type=str, default=None, required=False)
 @project_option(required=False)
-@click.option('--run-after/--no-run-after', 'run_after', default=True, help='Immediately run the uploaded program')
+@click.option('--run-after/--no-run-after', 'run_after', default=None, help='Immediately run the uploaded program')
 @click.option('-q', '--quirk', type=int, default=0)
 @click.option('--name', 'remote_name', type=str, default=None, required=False, help='Remote program name',
               cls=PROSOption, group='V5 Options')
-@click.option('--slot', default=1, show_default=True, type=click.IntRange(min=1, max=8), help='Program slot on the GUI',
+@click.option('--slot', default=None, type=click.IntRange(min=1, max=8), help='Program slot on the GUI',
               cls=PROSOption, group='V5 Options')
 @click.option('--program-version', default=None, type=str, help='Specify version metadata for program',
               cls=PROSOption, group='V5 Options', hidden=True)
@@ -27,11 +27,11 @@ def upload_cli():
               cls=PROSOption, group='V5 Options', hidden=True)
 @click.option('--ini-config', type=click.Path(exists=True), default=None, help='Specify a Program Configuration File',
               cls=PROSOption, group='V5 Options', hidden=True)
-@click.option('--run-screen/--execute', 'run_screen', default=True,
+@click.option('--run-screen/--execute', 'run_screen', default=None,
               cls=PROSOption, group='V5 Options', help='Open "run program" screen after uploading, instead of executing'
                                                        ' program. This option may help with controller connectivity '
                                                        'reliability and prevent robots from running off tables.')
-@click.option('--compress-bin/--no-compress-bin', 'compress_bin', cls=PROSOption, group='V5 Options', default=True,
+@click.option('--compress-bin/--no-compress-bin', 'compress_bin', cls=PROSOption, group='V5 Options', default=None,
               help='Compress the program binary before uploading.')
 @default_options
 def upload(path: Optional[str], project: Optional[c.Project], port: str, **kwargs):
@@ -58,7 +58,9 @@ def upload(path: Optional[str], project: Optional[c.Project], port: str, **kwarg
 
         # apply upload_options as a template
         options = dict(**project.upload_options)
-        options.update(kwargs)
+        for k, v in options:
+            if k not in kwargs.keys() or kwargs[k] is None:
+                kwargs[k] = v
         kwargs = options
 
         kwargs['target'] = project.target  # enforce target because uploading to the wrong uC is VERY bad
@@ -84,7 +86,10 @@ def upload(path: Optional[str], project: Optional[c.Project], port: str, **kwarg
         if kwargs['remote_name'] is None:
             kwargs['remote_name'] = os.path.splitext(os.path.basename(path))[0]
         kwargs['remote_name'] = kwargs['remote_name'].replace('@', '_')
+        kwargs['slot'] = kwargs['slot'] if kwargs['slot'] is not None else 1
         kwargs['slot'] -= 1
+        kwargs['run_after'] = kwargs['run_after'] if kwargs['run_after'] is not None else True
+        kwargs['run_screen'] = kwargs['run_screen'] if kwargs['run_screen'] is not None else True
         if kwargs['run_after'] and kwargs['run_screen']:
             kwargs['run_after'] = vex.V5Device.FTCompleteOptions.RUN_SCREEN
         elif kwargs['run_after'] and not kwargs['run_screen']:
```

Nenik, 2019-11-04:
Yes, that looks reasonable. I'll try to test tomorrow...

HotelCalifornia, 2019-11-16:
@Nenik did you ever get a chance to test Elliot's patch?

Nenik, 2019-11-19:
No, but trying now.
Not sure what's wrong (not skilled at Python at all), but the patch seems to break upload with upload_options populated.
With empty upload_options, upload seems to proceed normally, but once I configure a slot using:
```
        "upload_options": {
            "slot": 4
        }
```
the upload breaks:
```
ERROR - pros.cli.main:main - too many values to unpack (expected 2)
  File "..../pros-cli/pros/cli/upload.py", line 61, in upload
    for k, v in options:
ValueError: too many values to unpack (expected 2)
```
Did I apply the patch wrong?

Nenik, 2019-11-19:
It seems the options iterator would have to read:
```
+        for k, v in options.items():
```

lerker100, 2020-04-16:
I tried implementing this here.
https://github.com/lerker100/pros-cli/blob/78afda589d798262a9bc2f30435568b45beac30a/pros/cli/upload.py#L86-L94

I'm not quite sure if I implemented it the way that it should be and it has one bug because I couldn't find how to test if --slot was present.

HotelCalifornia, 2020-04-20:
> I'm not quite sure if I implemented it the way that it should be and it has one bug because I couldn't find how to test if --slot was present.

couple ways you could do it. one would be to just blindly access `kwargs['slot']` and catch a `KeyError` (i believe is the exception, might want to fact-check me there). another (probably better) way would be to do `slot = kwargs.get('slot', None)`, where `dict.get()` allows you to specify a default fallback value when the requested key is not present. then all you'd have to do is check `slot is not None`

closed 2021-01-24 by HotelCalifornia