purduesigbots/pros-cli #112: Controller DC's right after download

TaranMayer, 2019-10-27:
Right after I download with PROS the controller disconnects from the brain for about 5 seconds. This happens EVERY time. It does not happen with VEXcode. 

-Taran

theol0403, 2019-10-27:
This also happened to me the last time I worked with the robot via wireless downloading (April). This was on the firmware version before 3.2.0, using the current version of the CLI.

When I run wireless download, it switches to the download channel, downloads, and then completes. About 10-20 seconds later, the controller disconnects for about 5 seconds (I assume it is switching back channels). This is annoying when starting the program right after it finishes downloading, but then having the controller disconnect.

Here is the relevant conversation on the unofficial VEX discord server with multiple people chiming in with the same issue:
https://discordapp.com/channels/197777408198180864/198658294007463936/638120641408794634

HotelCalifornia, 2019-10-27:
not clear yet that this is necessarily a bug in what we're doing, but thank you for opening an issue so we can investigate

M888R, 2019-11-03:
summary of proposed solution:
create CLI flag to not switch off download channel on exit
https://github.com/purduesigbots/pros-cli/blob/d9c12776cd33ec1a7e5bed1145a407a05525b774/pros/serial/devices/vex/v5_device.py#L195-L198

HotelCalifornia, 2019-11-12:
May also want to tell upload not to switch channels when chaining commands like upload/terminal

nathan-moore, 2020-03-17:
There is some related discussion on the vex forum [here](https://www.vexforum.com/t/controller-disconnecting-at-10-seconds/78034/6). Most notably, it was pointed out that we never set did_switch to true, so we never switch back to the pits channel.

closed 2020-09-26 by HotelCalifornia