purduesigbots/pros-cli #117: prosv5 build-compile-commands UnicodeDecodeError

3038922, 2019-11-21:
windows --version :10 1909 (18363.476)
python --version: python3.7 or python3.8
pros-cli --version:pros-cli-v5      3.1.4
```
PS D:\ncrApiPlus> prosv5 build-compile-commands
Exception in thread Thread-1:
Traceback (most recent call last):
  File "c:\users\aresp\appdata\local\programs\python\python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "c:\users\aresp\appdata\local\programs\python\python37\lib\site-packages\pros\common\ui\__init__.py", line 180, in run
    for line in iter(self.pipeReader.readline, ''):
UnicodeDecodeError: 'gbk' codec can't decode byte 0x80 in position 10: illegal multibyte sequence
```
makefile:
```
# Set to 1 to enable hot/cold linking
USE_PACKAGE:=0

# Set this to 1 to add additional rules to compile your project as a PROS library template
IS_LIBRARY:=0
# TODO: CHANGE THIS!
LIBNAME:=libbest
VERSION:=1.0.0
```
I localized some makefile prompts。
Maybe that's the problem?Or I updated win10 to 1909？

nathan-moore, 2019-12-08:
This looks like a bug to me and something we should fix. Are you interested in opening a pr for your fix?

3038922, 2020-07-21:
```py
class _MachineOutputProgressBar(_click_ProgressBar):
    def __init__(self, *args, **kwargs):
        global _current_notify_value
        kwargs['file'] = open(os.devnull, 'w', encoding='UTF-8')
        self.notify_value = kwargs.pop('notify_value', _current_notify_value)
        super(_MachineOutputProgressBar, self).__init__(*args, **kwargs)
```

```py
class EchoPipe(threading.Thread):
    def __init__(self, err: bool = False, ctx: Optional[click.Context] = None):
        """Setup the object with a logger and a loglevel
        and start the thread
        """
        self.click_ctx = ctx or click.get_current_context(silent=True)
        self.is_err = err
        threading.Thread.__init__(self)
        self.daemon = False
        self.fdRead, self.fdWrite = os.pipe()
        self.pipeReader = os.fdopen(self.fdRead, encoding='UTF-8')
        self.start()
```

```
PS C:\Users\aresp\Desktop\usercode\test> prosv5 build-compile-commands
Creating cold package with okapilib,libm,libc,libpros [OK]
Stripping cold package  [DONE]
Section sizes:
   text    data     bss   total     hex filename
1.56MB  4.34KB  47.62MB  49.19MB 312e182 bin/cold.package.elf
Compiled src/main.cpp [OK]
Adding timestamp [OK]
Linking hot project with ./bin/cold.package.elf and okapilib,libm,libc,libpros [OK]
Section sizes:
   text    data     bss   total     hex filename
 1.13KB   4.00B   2.00B  1.14KB     486 bin/hot.package.elf
Creating cold package binary for VEX EDR V5 [DONE]
Creating bin/hot.package.bin for VEX EDR V5 [DONE]
Capturing metadata for PROS Editor...
```

PR #129 (2021-01-24): 🐛 Fix UnicodeDecodeError in UI
```
PS C:\Users\aresp\Desktop\usercode\test> prosv5 build-compile-commands
Creating cold package with okapilib,libm,libc,libpros [OK]
Stripping cold package  [DONE]
Section sizes:
   text    data     bss   total     hex filename
1.56MB  4.34KB  47.62MB  49.19MB 312e182 bin/cold.package.elf
Compiled src/main.cpp [OK]
Adding timestamp [OK]
Linking hot project with ./bin/cold.package.elf and okapilib,libm,libc,libpros [OK]
Section sizes:
   text    data     bss   total     hex filename
 1.13KB   4.00B   2.00B  1.14KB     486 bin/hot.package.elf
Creating cold package [... cut]
commit 5c6238e0 (2021-01-24): 🐛 Fix UnicodeDecodeError in UI (#129)

* fix #117

```
PS C:\Users\aresp\Desktop\usercode\test> prosv5 build-compile-commands
Creating cold package with okapilib,libm,libc,libpros [OK]
Stripping cold package  [DONE]
Section sizes:
   text    data     bss   total     hex filename
1.56MB  4.34KB  47.62MB  49.19MB 312e182 bin/cold.package.elf
Compiled src/main.cpp [OK]
Adding timestamp [OK]
Linking hot project with ./bin/cold.package.elf and okapilib,libm,libc,libpros [OK]
Section sizes:
   text    data     bss   total     hex filename
 1.13KB   4.00B   2.00B  1.14KB     486 bin/hot.package.elf
C [... cut]
closed 2021-01-24 by HotelCalifornia