purduesigbots/pros-cli #130: Is it time for a new release?

lerker100, 2020-09-15:
It's been a year and a half sense the last release of Pros CLI. With the new release of the Pros kernel, teams starting up for this season, and all the new features that have been developed. It seems like there needs to be an update. I for one have been using my own fork with some of the fixes and feature that aren't out sense April.

HotelCalifornia, 2020-09-15:
yes

lerker100, 2020-12-14:
Has this topic been discussed internally yet? It's been 4 months and teams really need these new features heading into the heavy competition season. I'm about to make a fork and implement all the new changes my self it's been so long.

lerker100, 2020-12-17:
@WillXuCodes I've compiled all the open PRs to a seperate git repo. I was wondering exactly what you all are looking for in a PR? So while I'm testing and merging the PRs I can keep that in mind.

lerker100, 2020-12-17:
@HotelCalifornia He asked for me to create a PR when I implement all the new changes. So I'm asking exactly how you all want that. I'm not sure if a PR with all the other PRs in one branch or what he was asking for?

WillXuCodes, 2020-12-17:
@lerker100 I may have originally misinterpreted what you were saying, I originally thought you said you worked on new features yourself and wanted to implement them on your own fork. No need for the PR then, but do note that some others do have their own CLI builds (with wireless terminal and other unreleased features) and are doing something similar to what you are planning to do.

kunwarsahni01, 2020-12-21:
Were working through CLI issues/PRs now.

closed 2020-12-21 by kunwarsahni01