purduesigbots/pros-cli #131: When updating to 3.3.0 main.cpp gets written if it doesn't exist

lerker100, 2020-09-21:
#### Expected Behavior:
Not to create this extraneous file breaking existing code.

#### Actual Behavior:
Creating main.cpp file in /src which my project doesn't use because it was started before the change from separate files.

#### Steps to reproduce:
Create a project
migrate your code to a different file
update pros to 3.3.0

#### System information:
Platform: V5
PROS Kernel Version: 3.3.0

#### Additional Information
I really don't like the change to one main.cpp file because of the large amount of code from autons making it hard to find the line that your looking for when debugging. And for newer users it's nice to already have that done for you so you don't have to learn how to do that on top of learning cpp and pros.

#### Screenshots/Output Dumps/Stack Traces
Nobody really needs to read this but putting it here anyways.
```
Compiled src/Chassis.cpp [OK]
Compiled src/Screen.cpp [OK]
Compiled src/autonomous.cpp [OK]
Compiled src/initialize.cpp [OK]
Compiled src/main.cpp [OK]
Compiled src/opcontrol.cpp [OK]
Adding timestamp [OK]
Linking hot project with ./bin/cold.package.elf and libc,libm,libpros,okapilib [ERRORS]
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: bin/main.cpp.o: in function `disabled':
main.cpp:(.text.disabled+0x0): multiple definition of `disabled'; bin/initialize.cpp.o:/usr/arm-none-eabi/include/c++/10.2.0/bits/unique_ptr.h:962: first defined here
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: bin/main.cpp.o: in function `competition_initialize':
main.cpp:(.text.competition_initialize+0x0): multiple definition of `competition_initialize'; bin/initialize.cpp.o:/usr/arm-none-eabi/include/c++/10.2.0/bits/shared_ptr_base.h:547: first defined here
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: bin/main.cpp.o: in function `autonomous':
/home/stephan/GitLab/2020-2021/./include/okapi/api/util/logging.hpp:185: multiple definition of `autonomous'; bin/autonomous.cpp.o:/usr/arm-none-eabi/include/c++/10.2.0/ext/new_allocator.h:150: first defined here
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: bin/main.cpp.o: in function `initialize':
/home/stephan/GitLab/2020-2021/src/main.cpp:25: multiple definition of `initialize'; bin/initialize.cpp.o:/home/stephan/GitLab/2020-2021/./include/okapi/api/util/logging.hpp:180: first defined here
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: bin/opcontrol.cpp.o: in function `opcontrol':
/home/stephan/GitLab/2020-2021/src/opcontrol.cpp:15: multiple definition of `opcontrol'; bin/main.cpp.o:/home/stephan/GitLab/2020-2021/src/main.cpp:76: first defined here
collect2: error: ld returned 1 exit status
Error: Failed to build
make: *** [common.mk:239: bin/hot.package.elf] Error 1
```

lerker100, 2020-09-21:
GitHub said that it transferred this issue but I don't see it on the pros repo. I had meant to put it there in the first place. Sorry.

HotelCalifornia, 2020-09-21:
it was transferred from purduesigbots/pros -> purduesigbots/pros-cli, as this is a CLI issue

HotelCalifornia, 2020-11-06:
we discussed this internally and came to the conclusion there's not really a good way to do this. @edjubuh maybe you want to weigh in here?

lerker100, 2020-11-06:
Couldn't you just leave out main.cpp from the tar ball? Or when updating not move files into src but move them when creating a new project?

HotelCalifornia, 2020-11-06:
> Couldn't you just leave out main.cpp from the tar ball?

no

> Or when updating not move files into src but move them when creating a new project?

potentially, but this would be (comparatively) a lot of work to add a feature with a very narrow use-case.

I'm not dismissing the idea out of hand, though. community feedback on this point is welcome

lerker100, 2020-11-06:
The reason I put an identical issue on the pros repo is one of you moved this issue from cli to pros and it didn't show up there.

HotelCalifornia, 2020-11-06:
> The reason I put an identical issue on the pros repo is one of you moved this issue from cli to pros and it didn't show up there.

again, we did the opposite. you originally opened this on the PROS repo and we transferred it here, to the PROS CLI repo. because, again, this is a CLI issue.

omegaStag, 2022-09-14:
this issue hasnt been relevant for a while and there is an easy workaround

closed 2022-09-14 by omegaStag