purduesigbots/pros-cli #132: Upload's run-after switch does nothing

Octogonapus, 2020-09-29:
#### Expected Behavior:
`prosv5 u --run-after` should cause the program to run after the upload completes successfully.

#### Actual Behavior:
`prosv5 u --run-after` uploads the program successfully like normal, but does not run the program after the upload is finished.

#### Steps to reproduce:
This is reproducible every time. Just run `prosv5 u --run-after` and observe it has the same behavior as `prosv5 u --no-run-after`.

#### System information:
Operating System:  Fedora 32

PROS Version: 3.1.4

#### Additional Information
This happens with hot/cold and monolith.

#### Screenshots/Output Dumps/Stack Traces

lerker100, 2020-11-28:
I use a bash alias on my linux PC's. `alias upload='prosv5 make; prosv5 upload; prosv5 v5 run'` but the command by it's self is `prosv5 v5 run`.

Octogonapus, 2020-11-28:
I am not sure what you are saying. I understand what `prosv5 v5 run` does. This issue is about `prosv5 u --run-after`, which does not work.

lerker100, 2020-11-28:
I thought you were just trying to upload after you compiled. which `prosv5 v5 upload` does. I now see that you are reporting that a specific setting in upload is currently broken.

BennyBot, 2021-10-17:
Currently, the `--run-screen` switch has priority over the `--run-after` switch. The proper way as of now to immediately run the program after uploading is `prosv5 u --run-after --execute`, which sets the run-after switch to true and the run-screen switch to false. I am working on making these switches easier to use for future releases 😃.