purduesigbots/pros-cli #134: Cross platform issues when uploading to different slots

MikeGstefan, 2020-11-01:
#### Expected Behavior:
When uploading to a different slot, the code remains in the other slot

#### Actual Behavior:
If the code in a different slot was uploaded from a different OS it is removed with the upload. When we upload into slot 2 (or any slot but 2 in this example) from a windows computer, if the other programs on the v5 were uploaded from a mac they get removed ex. a program in slot 1 was removed when trying to upload to slot 2

#### Steps to reproduce:
Upload from a different OS

#### System information:

Windows 10 and OS X 10.15.6

PROS Version: 3.3.1

#### Additional Information
Sometimes it will throw an error about not finding the device header, it does not do this every time.

#### Screenshots/Output Dumps/Stack Traces
https://cdn.discordapp.com/attachments/198658294007463936/772561313393147904/WhatsApp_Image_2020-11-01_at_14.13.52_1.jpeg   -> mac error with device header error

https://cdn.discordapp.com/attachments/198658294007463936/772559971476176946/WhatsApp_Image_2020-11-01_at_13.37.45.jpeg    -> windows error

https://cdn.discordapp.com/attachments/736766073608732706/772575758597750794/unknown.png   ->windows error with device header

edjubuh, 2020-11-02:
I think I read two issues here:

1. Sometimes device header isn't found on both mac and windows.
  This needs to be investigated further.

2. Uploading program A and program B from one OS, and later trying to upload program B from another OS causes program A to be removed.
  This may be intended behavior depending on storage available on the device. PROS library and the associated user programs may be removed if V5 reports not enough space. The PROS library is host-specific today and can't be reused if you compile the same program on a different computer.

MikeGstefan, 2020-11-02:
> I think I read two issues here:
>
>
>
> 1. Sometimes device header isn't found on both mac and windows.
>
>   This needs to be investigated further.
>
>
>
> 2. Uploading program A and program B from one OS, and [...]

1. I only remember encountering this error when Im getting the other one as well, I have yet to see it independently.

2. We were trying to upload completely different programs from the other OS when we encountered the error.

HotelCalifornia, 2020-11-02:
> 2\. We were trying to upload completely different programs from the other OS when we encountered the error.

yeah the problem elliot noted is that the cold package (libpros.a, etc) gets compiled per-host in this case. so compiling from different machines does technically produce different cold packages, which as you've seen can trigger the ensure_library_space behavior

lerker100, 2020-11-28:
I also have noticed this issue when deleting surten files from a pros project. I just assumed that it was a vex problem.