purduesigbots/pros-cli #135: Build fails in both cli and in atom

skim3650, 2020-11-25:
#### Expected Behavior:

I tried to build the project with prosv5 make
#### Actual Behavior:

Compiled src/main.cpp [ERRORS]
/bin/sh: intercept-c++: command not found
make: *** [common.mk:270: bin/main.cpp.o] Error 127
Error: Failed to build

#### Steps to reproduce:

this is what happens when I build with cli, on atom it just says 
make: nothing to be done for 'quick'.

#### System information:
Operating System:  
Windows 10 
PROS Version: 
3.1.4
#### Additional Information

I got an error before that said
ERROR - pros.cli.conductor:new_project - [WinError 2] The system cannot find the file specified
  File "C:\hostedtoolcache\windows\Python\3.6.4\x64\lib\subprocess.py", line 709, in __init__
  File "C:\hostedtoolcache\windows\Python\3.6.4\x64\lib\subprocess.py", line 997, in _execute_child
FileNotFoundError: [WinError 2] The system cannot find the file specified
#### Screenshots/Output Dumps/Stack Traces

lerker100, 2020-11-28:
I'm not super familiar with this exact error. But a couple of things I would try. I'm pretty sure that your not in the right spot or missing a dependency.

1. Open pros atom using the windows file explorer in your project folder.
2. Creating a new project and building it. Without modifying the default project. 
3. Uninstall Pros and Pros atom and reinstall.

HotelCalifornia, 2020-11-28:
Why does the output mention /bin/sh if you're using windows? are you using the windows subsystem for linux?

stevejluke2, 2021-04-13:
Same on Windows 10, except the error is about gc++:

c:\Projects\VexVRC\PROS\TestRos>pros make
Creating cold package with libc,libm,libpros,okapilib Error: Failed to build
[ERRORS]
/bin/sh: line 1: arm-none-eabi-g++: command not found
make: *** [common.mk:229: bin/cold.package.elf] Error 127

Why /bin/sh?  No idea.  This is through command prompt, PowerShell, and Atom.  Multiple installs.

Sleuth56, 2021-04-13:
@stevejluke2 it looks like you just don't have g++ installed.

HotelCalifornia, 2021-04-13:
> @stevejluke2 it looks like you just don't have g++ installed.

not quite

> Same on Windows 10, except the error is about gc++:
> 
> c:\Projects\VexVRC\PROS\TestRos>pros make
> Creating cold package with libc,libm,libpros,okapilib Error: Failed to build
> [ERRORS]
> /bin/sh: line 1: arm-none-eabi-g++: command not found
> make: *** [common.mk:229: bin/cold.package.elf] Error 127

you should read the section on known issues and workarounds for windows on [the release page](https://github.com/pros-cli/releases/3.2.0), which address this issue specifically

stevejluke2, 2021-04-14:
Yes, the instructions on that page work. I missed that those instructions were for more than just the version downloaded during a specific period of time. But setting the PATH variable to include g++ did the job. 

Thanks

closed 2021-04-14 by HotelCalifornia