purduesigbots/pros-cli #136: Kernel update doesn't cause Cold section to be evicted

nathan-moore, 2020-12-02:
#### Expected Behavior:
When you update the kernel, the cold section is evicted and re-uploaded.

#### Actual Behavior:

Bad things. Prefetch errors is one I saw, and would explain why we see them occasionally.

#### Steps to reproduce:

On a brain with a old onboard library, update the pros kernel. It will not upload the new cold section.

#### System information:
Operating System:  
All
PROS Version:  latest (probably all)

#### Additional Information

See messages from drcheddar#4883 on discord for the latest occurrence I've seen of this.

HotelCalifornia, 2020-12-02:
cc @edjubuh

lerker100, 2020-12-14:
Doesn't the CLI trigger a complete rebuild of a project after preforming an update? I'm sure that knowing what it isn't doing would be helpful for whoever tries to work this out.

nathan-moore, 2020-12-15:
>Doesn't the CLI trigger a complete rebuild of a project after preforming an update? I'm sure that knowing what it isn't doing would be helpful for whoever tries to work this out.

Yes, and a backwards way this is part of the problem. It's built with some code layout and is run with an older one.

omegaStag, 2022-09-14:
intentional design

closed 2022-09-14 by omegaStag