purduesigbots/pros-cli #139: ProsV5 Make returns error on latest Archlinux

Geometrically, 2020-12-12:
#### Expected Behavior:
Project is built and made

#### Actual Behavior:
This output happens
```
Traceback (most recent call last):
  File "/home/geometrically/.local/bin/prosv5", line 33, in <module>
    sys.exit(load_entry_point('pros-cli-v5', 'console_scripts', 'prosv5')())
  File "/home/geometrically/.local/bin/prosv5", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "/usr/lib/python3.9/importlib/metadata.py", line 524, in distribution
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.9/importlib/metadata.py", line 187, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: pros-cli-v5
```
#### Steps to reproduce:
Install prosV5 on latest arch packages, and try to make a new projec

#### System information:
Platform: V5
PROS Kernel Version: 3.3.1

#### Additional Information
Arch Linux

#### Screenshots/Output Dumps/Stack Traces
```
Traceback (most recent call last):
  File "/home/geometrically/.local/bin/prosv5", line 33, in <module>
    sys.exit(load_entry_point('pros-cli-v5', 'console_scripts', 'prosv5')())
  File "/home/geometrically/.local/bin/prosv5", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "/usr/lib/python3.9/importlib/metadata.py", line 524, in distribution
    return Distribution.from_name(distribution_name)
  File "/usr/lib/python3.9/importlib/metadata.py", line 187, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: pros-cli-v5
```

HotelCalifornia, 2020-12-12:
how did you install? using the wheel file?

Geometrically, 2020-12-12:
Yep

HotelCalifornia, 2020-12-12:
maybe a silly question but have you tried just reinstalling the CLI from the wheel?

HotelCalifornia, 2020-12-12:
if that doesn't work, you might want to purge the pip dependencies and then reinstall again

Geometrically, 2020-12-12:
How should I do both? I tried reinstalling but I may have done so incorrectly. Is this a bug with Python 3.9?

lerker100, 2020-12-14:
Can you list all commands you used? Starting from the first install command and ending with the project creation command. (the one throwing the error)

closed 2021-01-07 by Geometrically