purduesigbots/pros-cli #140: Remove links to the docs pages in the Help Messages

WillXuCodes, 2020-12-15:
With updates to the docs, these links are no longer necessary.

WillXuCodes, 2021-06-07:
Somewhere along the line this was either done or became irrelevant as well.

closed 2021-06-07 by WillXuCodes

HotelCalifornia, 2021-06-07:
that's not entirely true, there dead links still present in at least the conductor commands, and possibly in other places as well

WillXuCodes, 2021-06-07:
> that's not entirely true, there dead links still present in at least the conductor commands, and possibly in other places as well

Well I checked the link in the conductor command, and it still sends you to the conductor page (albeit with less information than it used to have), but it's still an active link that's appropriate for the command.