purduesigbots/pros-cli #143: Broken in Python 3.9.1

Geometrically, 2021-01-07:
#### Expected Behavior:
Pros CLI works as normal

#### Actual Behavior:
Pros CLI errors out, and no commands are able to be executed

#### Steps to reproduce:
Install Python 3.9.1, run pros

#### System information:
Operating System: Manjaro

PROS Version: 3.1.4

#### Screenshots/Output Dumps/Stack Traces
Traceback (most recent call last):
  File "/usr/bin/prosv5", line 5, in <module>
    from pros.cli.main import main
ModuleNotFoundError: No module named 'pros'

HotelCalifornia, 2021-01-07:
okay! I finally spun up an arch instance to see what's going on here. two things:

1. how did you attempt to install the CLI?
2. check the output of `python -m site` against your `PATH`. it looks like if you specify `--user` (as you should), the package gets installed to `~/.local/lib/python3.9/site-packages`, no part of which appears to be on `PATH` by default.

HotelCalifornia, 2021-01-07:
this will probably require an update in [purduesigbots/pros-docs](https://github.com/purduesigbots/pros-docs)

Geometrically, 2021-01-07:
I installed the CLI through the instructions here: (https://pros.cs.purdue.edu/v5/getting-started/linux.html). I ran the command, but with `sudo` and no --user flag so I could install it for all users. I switched to Manjaro, and this started happening right after I upgraded my system which installed Python 3. I had it working completely before the Manjaro update.

HotelCalifornia, 2021-01-07:
> I installed the CLI through the instructions here: (https://pros.cs.purdue.edu/v5/getting-started/linux.html). I ran the command, but with `sudo` and no --user flag so I could install it for all [...]

well alright. i think i asked you in the previous issue you opened and got no response---have you tried simply re-running the install command?

if you have, and it still doesn't work, figure out where the binary was actually placed (`python -m site` might provide some good places to start looking) and verify that's on PATH

Geometrically, 2021-01-08:
Sorry, I didn't reply to the previous issue since I fixed the issue since I got a new computer and switched to manjaro.

I'll rerun the command right now.

It worked LOL. The reinstall command worked. Thanks!

HotelCalifornia, 2021-01-08:
okay, i'm not really sure what changed to break it but i'm glad that worked

Geometrically, 2021-01-09:
I can't upload programs for some reason. 
```
Traceback (most recent call last):
  File "/home/geometrically/.local/bin/prosv5", line 5, in <module>
    from pros.cli.main import main
ModuleNotFoundError: No module named 'pros'```
Is the error. All other commands work @HotelCalifornia

closed 2021-01-09 by Geometrically