purduesigbots/pros-cli #157: PROS not Identifying stdio

Anivishy, 2021-03-27:
Expected Behavior:
We expected the code to build wihtout any errors as it was working fine before we updated to the new version of pros.

Actual Behavior:
PROS failed to identify stdio, which caused the code to not be able to build/upload.

Steps to reproduce:
This error simply comes up whenever we try to use PROS. We have tried adding a path, uninstalling and reinstalling as well as repairing it but nothing has seemed to work.

System information:
Operating System:
Windos
PROS Version: 
3.2.0

Additional Information
Screenshots/Output Dumps/Stack Traces

WillXuCodes, 2021-06-07:
Do you have a screenshot of the error? Also, the newest version of PROS is version 3.2.1 so I recommend updating. Furthermore, if you haven't updated your Kernel to 3.4.0 please do so as well.

WillXuCodes, 2021-06-07:
Also please make a post on VEXForum or the Unofficial VEX Discord for technical help, the issues section here is for issues with the PROS code itself.

closed 2021-06-07 by WillXuCodes