purduesigbots/pros-cli #159: ERROR - pros.cli.upload:upload - Recieved length doesn't match 33 (got 37)

micahsgilbert, 2021-04-07:
#### Expected Behavior:
The program uploads normally after running `prosv5 upload`

#### Actual Behavior:
After running `prosv5 upload`, i get the error
```
$ prosv5 upload
ERROR - pros.cli.upload:upload - Received length doesn't match 33 (got 37)
```
The full dump is below

#### Steps to reproduce:
I can't tell why this is happening; it seems to only happen with specific v5 brains.

#### System information:
Ubuntu 20.04.02 LTS

PROS Version: 3.1.4

#### Screenshots/Output Dumps/Stack Traces
```
$ prosv5 upload
ERROR - pros.cli.upload:upload - Received length doesn't match 33 (got 37)
TX:C9 36 B8 47 56 22 00 60 FC   RX:AA 55 56 29 22 76 00 01 00 0D 00 01 00 00 24 01 00 0D 35 06 00 00 00 00 28 18 BE 00 03 02 00 21 00 00 00 01 00 1B 00 01 01 00 07 72 22
    raise VEXCommError("Received length doesn't match {} (got {})".format(rx_length, len(msg)), msg)
pros.serial.devices.vex.comm_error.VEXCommError: Received length doesn't match 33 (got 37)
TX:C9 36 B8 47 56 22 00 60 FC   RX:AA 55 56 29 22 76 00 01 00 0D 00 01 00 00 24 01 00 0D 35 06 00 00 00 00 28 18 BE 00 03 02 00 21 00 00 00 01 00 1B 00 01 01 00 07 72 22
We detected something went wrong! Do you want to send a report? [y/N]:
```

micahsgilbert, 2021-04-07:
I've tried:
- Clearing the programs on the v5 brain
- Factory resetting the brain
- Uploading a fresh PROS project - gives the same error

WillXuCodes, 2021-04-07:
Hi Micah, 

You need to upgrade the CLI itself, not the kernel (which is what prosv5 conductor upgrade does). Only the latest version of PROS supports the newest vexOS version.

https://www.vexforum.com/t/pros-cli-3-2-0-wireless-terminal-important/89429?u=hotel

- The PROS Dev Team

closed 2021-04-07 by WillXuCodes

micahsgilbert, 2021-04-08:
Fixed, thanks!