purduesigbots/pros-cli #168: ERROR - pros.cli.upload:upload - Couldn't find the response header in the device response after 0.1 s. Got  but was expecting aa55

3038922, 2021-08-02:
win10 20h1

This happens only on my own computer. The student computers are all normal. I don’t know if my computer is USB3.1?

```
PS C:\Users\admin\Desktop\usercode\Examination_Guide> pros --version
pros, version 3.2.2
```

```
PS C:\Users\admin\Desktop\usercode\Examination_Guide> python --version 
Python 3.8.9
```

```
PS C:\Users\admin\Desktop\usercode\Examination_Guide> clang --version
clang version 11.1.0 (https://github.com/llvm/llvm-project.git 1fdec59bffc11ae37eb51a1b9869f0696bfd5312)
Target: x86_64-pc-windows-msvc
Thread model: posix
InstalledDir: c:\ncrRobotics\llvm\Release\bin
```

```
PS C:\Users\admin\Desktop\usercode\Examination_Guide> pros lsusb
VEX EDR V5 System Ports:
COM9 - USB 串行设备 (COM9)
VEX EDR V5 User ports:
COM10 - USB 串行设备 (COM10)
There are no connected VEX EDR Cortex Microcontroller Ports
```
```
S C:\Users\admin\Desktop\usercode\Examination_Guide> pros mut 
make: Nothing to be done for 'quick'.
ERROR - pros.cli.upload:upload - Couldn't find the response header in the device response after 0.1 s. Got  but was expecting aa55
  File "c:\users\admin\appdata\local\programs\python\python38\lib\site-packages\pros\serial\devices\vex\vex_device.py", line 75, in _rx_packet
    raise IOError(f"Couldn't find the response header in the device response after {timeout} s. "
OSError: Couldn't find the response header in the device response after 0.1 s. Got  but was expecting aa55
We detected something went wrong! Do you want to send a report? [y/N]: n
Not sending bug report.
```

3038922, 2021-08-02:
```
    def _rx_packet(self, timeout: Optional[float] = None) -> Dict[str, Union[Union[int, bytes, bytearray], Any]]:
        # Optimized to read as quickly as possible w/o delay
        start_time = time.time()
        response_header = bytes([0xAA, 0x55])
        response_header_stack = list(response_header)
        rx = bytearray()
        if timeout is None:
            timeout = self.default_timeout
        while (len(rx) > 0 or time.time() - start_time < timeout) and len(response_header_stack) > 0:
            b = self.port.read(1)
            if len(b) == 0:
                continue
            b = b[0]
            if b == response_header_stack[0]:
                response_header_stack.pop(0)
                rx.append(b)
            else:
                logger(__name__).debug("Tossing rx ({}) because {} didn't match".format(bytes_to_str(rx), b))
                response_header_stack = bytearray(response_header)
                rx = bytearray()
        if not rx == bytearray(response_header):
            raise IOError(f"Couldn't find the response header in the device response after {timeout} s. "
                          f"Got {rx.hex()} but was expecting {response_header.hex()}")
        rx.extend(self.port.read(1))
        command = rx[-1]
        rx.extend(self.port.read(1))
        payload_length = rx[-1]
        if command == 0x56 and (payload_length & 0x80) == 0x80:
            logger(__name__).debug('Found an extended message payload')
            rx.extend(self.port.read(1))
            payload_length = ((payload_length & 0x7f) << 8) + rx[-1]
        payload = self.port.read(payload_length)
        rx.extend(payload)
        return {
            'command': command,
            'payload': payload,
            'raw': rx
        }
```

mattfbacon, 2021-11-16:
I get this if the programming module is not connected to a joystick, or if the joystick is not connected to the microcontroller.

closed 2023-11-06 by 3038922