purduesigbots/pros-cli #171: pros make giving python error

lolgeny, 2021-09-23:
#### Expected Behavior:
I ran `pros make`, and expected a successful build.

#### Actual Behavior:
I got this output:
```
$ pros make
ERROR - __main__:main - [WinError 3] The system cannot find the path specified
  File "C:\hostedtoolcache\windows\Python\3.7.9\x64\lib\subprocess.py", line 800, in __init__
  File "C:\hostedtoolcache\windows\Python\3.7.9\x64\lib\subprocess.py", line 1207, in _execute_child
FileNotFoundError: [WinError 3] The system cannot find the path specified
We detected something went wrong! Do you want to send a report? [y/N]:
```
Additionally, no matter if I send a bug report or not it freezes the terminal and becomes unresponsive to ctrl-c, ctrl-d, and ctrl-z

#### Steps to reproduce:
- Make a new project
- `pros make` or `pros build`
Unfortunately I don't know more than that. I've built successfully on my machine before.

#### System information:
Platform: v5
PROS Kernel Version: 3.5.2

#### Additional Information
Windows, pros 3.2.2. Other non-build cli commands seem to work.

#### Screenshots/Output Dumps/Stack Traces
`pros make --debug --verbose`
```
$ pros make --verbose --debug
INFO - pros.cli.common:callback - Debugging messages enabled
DEBUG - pros:callback - CLI Version: 3.2.2
INFO - pros.conductor.project:find_project - Found Project Path: C:\Users\me\Documents\coding\video\project.pros
INFO - pros.conductor.project:find_project - Found Project Path: C:\Users\me\Documents\coding\video\project.pros
DEBUG - pros.config.config:__init__ - Opening C:\Users\me\Documents\coding\video\project.pros (Project)
DEBUG - pros.config.config:__init__ - Opening C:\Users\me\AppData\Roaming\PROS\cli.pros (CliConfig)
ERROR - __main__:main - [WinError 3] The system cannot find the path specified
Traceback (most recent call last):
  File "pros/cli/main.py", line 47, in main
  File "C:\hostedtoolcache\windows\Python\3.7.9\x64\lib\site-packages\click\core.py", line 697, in main
  File "C:\hostedtoolcache\windows\Python\3.7.9\x64\lib\site-packages\click\core.py", line 1066, in invoke
  File "C:\hostedtoolcache\windows\Python\3.7.9\x64\lib\site-packages\click\core.py", line 895, in invoke
  File "C:\hostedtoolcache\windows\Python\3.7.9\x64\lib\site-packages\click\core.py", line 535, in invoke
  File "D:\a\1\s\pros\cli\build.py", line 23, in make
  File "D:\a\1\s\pros\conductor\project\__init__.py", line 386, in compile
  File "D:\a\1\s\pros\conductor\project\__init__.py", line 230, in make
  File "C:\hostedtoolcache\windows\Python\3.7.9\x64\lib\site-packages\sentry_sdk\integrations\stdlib.py", line 193, in sentry_patched_popen_init
  File "C:\hostedtoolcache\windows\Python\3.7.9\x64\lib\subprocess.py", line 800, in __init__
  File "C:\hostedtoolcache\windows\Python\3.7.9\x64\lib\subprocess.py", line 1207, in _execute_child
FileNotFoundError: [WinError 3] The system cannot find the path specified
We detected something went wrong! Do you want to send a report? [y/N]: 
```
(name changed to `me` for privacy reasons)

lolgeny, 2021-09-23:
Sorry I should've read the description before posting it, thanks for moving

nathan-moore, 2021-09-23:
If you go to %PROS_TOOLCHAIN%/bin, are there any files? Is make.exe there? Is make on your path a different way?

WillXuCodes, 2021-10-11:
For the sake of organization (we have a bunch of new recruits and I'm trying to make this less confusing), I'm going to close this as this can be fixed by setting the PROS_TOOLCHAIN variable properly or running the repair setting on the installer. Also closing this due to inactivity.